s60sc/ESP32-CAM_MJPEG2SD #395: Error webServer.cpp: failed to send /data/MJPEG2SD.htm to browser

ronp2001, 2024-03-26:
I apologize in advance because this is the first time i am working with an ESP32, so i do not know if i'm providing too little information in this issue:

I can't seem to be able to get to the setup page.
i've tried 2 seperate ESP32-CAM boards both get the same error.

Serial monitor output:
```[00:00:03.389 logSetup] =============== ESP-CAM_MJPEG 9.6.2 ===============
[00:00:03.400 WARN initBrownout] Brownout occurred due to inadequate power supply 
[00:00:03.400 logSetup] Compiled with arduino-esp32 v2.0.14
[00:00:03.410 printResetReason] Software reset due to brownout
[00:00:03.411 printWakeupReason] Wakeup by reset
[00:00:03.476 infoSD] SD card type SDHC, Size: 58.5GB
[00:00:03.477 listFolder] Sketch size 1.5MB
[00:00:03.777 listFolder] File: /data/common.js, size: 28604 bytes
[00:00:03.780 listFolder] File: /data/configs.txt, size: 5661 bytes
[00:00:03.782 listFolder] File: /data/MJPEG2SD.htm, size: 95KB
[00:00:03.793 listFolder] SD_MMC: 640KB used of 58.5GB
[00:00:03.794 loadConfig] Load config
[00:00:04.019 updateAppStatus] Enabling motion detection
[00:00:04.031 setup] PSRAM size: 8.0MB
[00:00:04.229 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:04.487 setWifiSTA] Wifi Station IP from DHCP
[00:00:04.560 WARN startWifi] SSID  not available, use AP 
[00:00:04.563 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_F81F7615B80C
[00:00:04.563 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_F81F7615B80C started, use 'http://192.168.4.1' to connect
[00:00:07.786 startPing] Started ping monitoring - On
[00:00:07.791 startWebServer] Starting web server on port: 80
[00:00:07.791 startWebServer] Remote server certificates not checked
[00:00:07.792 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:07.802 prepUpload] File uploads will use FTP server
[00:00:07.802 prepTelegram] Telegram not being used
[00:00:07.850 updateAppStatus] Enabling motion detection
[00:00:08.011 prepRecording] To record new AVI, do one of:
[00:00:08.011 prepRecording] - press Start Recording on web page
[00:00:08.012 prepRecording] - move in front of camera
 
[00:00:08.022 prepRecording] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[00:00:08.032 checkMemory]  Free: heap 103548, block: 77812, min: 103532, pSRAM 1490751
[00:00:08.033 loop] =============== Total tasks: 16 ===============

[00:00:10.269 isNight] Night time
[00:00:12.788 resetWatchDog] WatchDog started using task: ping
[00:00:16.256 onWiFiEvent] WiFi AP client connection
[00:00:17.983 ERROR @ webServer.cpp:36] Failed to send /data/MJPEG2SD.htm to browser 
[00:01:08.919 onWiFiEvent] WiFi AP client disconnection```

s60sc, 2024-03-26:
`[00:00:03.400 WARN initBrownout] Brownout occurred due to inadequate power supply `
wifi uses a lot of power

ronp2001, 2024-03-26:
ah okay thank you, i kinda figured cause i plugged it directly into my pc the usb port would've supplied 5V. but after measuring it with my multi meter it seems to be supplying ~4V instead. so i'll see if i can find a more stable power source of 5V

ronp2001, 2024-03-28:
Got it working after connecting it to my laptop. and using one of those usb to breadboard adapters to grab ~5V and powering it that way. afterwards using my laptops wi-fi card to connect to the network the esp creates. before i was using my mobile phone to try and reach the webpage.

closed 2024-03-28 by ronp2001