s60sc/ESP32-CAM_MJPEG2SD #397: [ISSUE] 16Mhz XCLK EDMA Mode not working

rjsachse, 2024-03-31:
Having an issue setting the xclk to 16Mhz, other frequency works. 16Mhz enables edma mode on esp32s3 to write frame buffer direct to psram to get faster fps.

s60sc, 2024-03-31:
[issue](https://github.com/espressif/esp32-camera/issues/574) with camera driver not app.

Note that edma mode is only useful for high res bitmaps due to their very large size. Of no use to this app.

rjsachse, 2024-04-01:
Ok thanks for the info So can't use 16Mhz XCLK. I was trying to get better fps on video streaming, only getting about 10fps but recording I can get 25fps. If recording and streaming at same time I get 25fps recording but only about 10fps streaming. Is there a way to increase fps for streaming.

s60sc, 2024-04-01:
Streaming dependent on wifi conn quality - fit better antenna.

rjsachse, 2024-04-01:
Ok I will try a few things, wifi connection is great and have 8dbi antenna installed.

closed 2024-04-06 by s60sc