s60sc/ESP32-CAM_MJPEG2SD #405: inbrowser flash improv wifi config

ldijkman, 2024-04-07:
would be nice if there is a 
inbrowser flash improv wifi config
version

https://ldijkman.github.io/async-esp-fs-webserver/

s60sc, 2024-04-07:
You can do this with **[WebDAV](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=streams%20may%20stutter.-,WebDAV,-A%20simple%20WebDAV)**

mDNS is only enabled for ESP32S3, but you can enable it for ESP32 (with consequences) by editing `utils.cpp` to remove the compiler directive:
```
#if CONFIG_IDF_TARGET_ESP32S3
  setupMdnsHost(); // not on ESP32 as uses 6k of heap
#endif
```

ldijkman, 2024-04-07:
http://esp32cam.local

http://garden.local/bulbs.html
- auto list all other devices in network iframed (bulb.html)
- since espcam does not have a bulb.html no page
- http://esp32cam.local/bulb.html should show a video preview maybe
- maybe i should use cards.html  and card.html
- 
[image]

[image]

ldijkman, 2024-04-07:
added bilb.html and index.html to sd card
with the same contents as main index htm file
but it does not handle it
not found
and then cannot load main ip adress anymore after bad url in adress bar

server.on or something must fallback to main if not found

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

[00:00:00.480 logSetup] Setup RAM based log, size 7168, starting from 6690

[00:00:00.480 logSetup] =============== ESP-CAM_MJPEG 9.7 ===============
[00:00:00.491 logSetup] Compiled with arduino-esp32 v2.0.14
[00:00:00.491 printResetReason] Power on reset
[00:00:00.501 printWakeupReason] Wakeup by reset
[00:00:00.556 infoSD] SD card type SDHC, Size: 7.4GB
[00:00:00.557 listFolder] Sketch size 1.6MB
[00:00:00.883 listFolder] File: /data/common.js, size: 28604 bytes
[00:00:00.894 listFolder] File: /data/MJPEG2SD.htm, size: 95KB
[00:00:00.905 listFolder] File: /data/configs.txt, size: 5691 bytes
[00:00:00.916 listFolder] File: /data/log.txt, size: 56KB
[00:00:00.927 listFolder] File: /data/index.html, size: 95KB
[00:00:00.940 listFolder] File: /data/bulb.html, size: 95KB
[00:00:00.941 listFolder] SD_MMC: 101.2MB used of 7.4GB
[00:00:00.942 loadConfig] Load config
[00:00:01.246 updateAppStatus] Disabling motion detection
[00:00:01.248 setupLamp] Setup Lamp Led for ESP32 Cam board
 
[00:00:01.272 remote_log_init_SD] Opened SD file for logging
[00:00:01.280 setup] PSRAM size: 8.0MB
[00:00:01.477 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:01.739 setWifiSTA] Wifi Station IP from DHCP
.......[00:00:04.847 onWiFiEvent] WiFi Station connection to Bangert_30_Andijk, using hostname: ESP32CAM
...[00:00:06.370 onWiFiEvent] Wifi Station IP, use 'http://10.10.100.122' to connect
[00:00:10.263 startWifi] Wifi stats for Bangert_30_Andijk - signal strength: -60 dBm; Encryption: WPA2_PSK; channel: 11
[00:00:10.265 startPing] Started ping monitoring - Off
[00:00:10.280 startWebServer] Starting web server on port: 80
[00:00:10.280 startWebServer] Remote server certificates not checked
[00:00:10.288 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:10.292 prepUpload] File uploads will use FTP server
[00:00:10.292 setupLamp] Setup Lamp Led for ESP32 Cam board
[00:00:10.303 prepTelegram] Telegram not being used
[00:00:10.303 resetWatchDog] WatchDog started using task: ping
[00:00:10.322 getLocalNTP] Using NTP server: pool.ntp.org
[00:00:10.324 WARN getLocalNTP] Not yet synced with NTP 
[00:00:10.347 updateAppStatus] Disabling motion detection
 
[00:00:11.475 remote_log_init_SD] Opened SD file for logging
 
[00:00:12.510 remote_log_init_SD] Opened SD file for logging
[00:00:12.591 prepRecording] To record new AVI, do one of:
[00:00:12.592 prepRecording] - press Start Recording on web page
 
[00:00:12.603 prepRecording] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[00:00:12.604 checkMemory]  Free: heap 101524, block: 77812, min: 99420, pSRAM 1490679
MDNS responder started.
You should be able to connect with address http://[32mesp32cam.local/
[0m[00:00:12.629 loop] =============== Total tasks: 17 ===============

[14:49:13.390 wsHandler] Websocket connection: 53
 
[14:49:14.458 remote_log_init_SD] Opened SD file for logging
[14:49:36.957 wsHandler] Websocket connection: 53
 
[14:49:38.013 remote_log_init_SD] Opened SD file for logging
[14:49:38.819 getLocalNTP] Using NTP server: pool.ntp.org
[14:49:38.820 showLocalTime] Got current time from NTP: 07/04/2024 14:49:38 with tz: GMT0
[14:49:43.843 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm 
[14:49:43.892 showStream] MJPEG: 21 frames, total 326KB in 5.1s @ 4.1fps
[14:49:52.150 showStream] MJPEG: 7 frames, total 103KB in 1.7s @ 4.2fps
[14:50:08.824 setAlarm] Alarm scheduled at 08/04/2024 01:00:00
[14:50:09.750 getExtIP] External IP: 83.80.183.132
[14:50:09.753 doAppPing] Daily rollover
[14:54:08.819 isNight] Night time
[14:55:36.676 showStream] MJPEG: 1370 frames, total 20.6MB in 342.4s @ 4.0fps
[14:55:36.688 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm 
[14:55:40.552 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm 
```

ldijkman, 2024-04-07:
main ino with mdns 
for  http://esp32cam.local

```

/*
* Capture ESP32 Cam JPEG images into a AVI file and store on SD
* AVI files stored on the SD card can also be selected and streamed to a browser as MJPEG.
*
* s60sc 2020 - 2024
*/

#include "appGlobals.h"
#include <ESPmDNS.h>
// http://esp32cam.local
const char* hostname = "esp32cam";  // .local is added by esp32 mdns   http://esp32cam.local
String myhostname = hostname;

void setup() {  
  logSetup();
  // prep SD card storage
  if (startStorage()) {
    // Load saved user configuration
    if (loadConfig()) {
      // initialise camera
      if (psramFound()) {
        LOG_INF("PSRAM size: %s", fmtSize(esp_spiram_get_size()));
        if (esp_spiram_get_size() > 3 * ONEMEG) prepCam();
        else snprintf(startupFailure, SF_LEN, STARTUP_FAIL "Insufficient PSRAM for app");
      }
      else snprintf(startupFailure, SF_LEN, STARTUP_FAIL "Need PSRAM to be enabled");
    }
  }
  
#ifdef DEV_ONLY
  devSetup();
#endif

  // connect wifi or start config AP if router details not available
  startWifi();

  startWebServer();
  if (strlen(startupFailure)) LOG_WRN("%s", startupFailure);
  else {
    // start rest of services
    startSustainTasks(); 
#if INCLUDE_SMTP
    prepSMTP(); 
#endif
#if INCLUDE_FTP_HFS
    prepUpload();
#endif
    prepPeripherals();
#if INCLUDE_MIC
    prepMic(); 
#endif
#if INCLUDE_TELEM
    prepTelemetry();
#endif
#if INCLUDE_TGRAM
    prepTelegram();
#endif
    prepRecording(); 
    checkMemory();
  } 

    if (MDNS.begin(myhostname)) {
    Serial.println("MDNS responder started.");
    Serial.print("You should be able to connect with address http://");
    Serial.print("\033[32m"); // Set green color (other color codes available)

    Serial.print(myhostname);
    Serial.println(".local/");
    MDNS.addService("http", "tcp", 80);
    MDNS.setInstanceName(myhostname);  // Change "new-service-name" to your desired name
    Serial.print("\033[0m"); // Reset color
  } else {
    Serial.println("Error setting up MDNS responder!");
  }
}

void loop() {
  // confirm not blocked in setup
  LOG_INF("=============== Total tasks: %u ===============\n", uxTaskGetNumberOfTasks() - 1);
  delay(1000);
  vTaskDelete(NULL); // free 8k ram
}
```

s60sc, 2024-04-07:
As previously stated the app already supports mdns, you dont need to change the .ino

If you want to access `bulb.html`, 
in `appGlobals.h` change
`#define INDEX_PAGE_PATH DATA_DIR "/MJPEG2SD" HTML_EXT`
to
`#define INDEX_PAGE_PATH DATA_DIR "/bulb" HTML_EXT`

ldijkman, 2024-04-07:
thanks for your response 
but this does not do what i want

i will talk to chatgpt gemeini copilot or blackbox

would like to serve any html file from sdcard
would like to add inbrowser ace editor html on sdcard
so that it would be easier to edit the html files directly on sdcard webserver inbrowser

greet luberth

rjsachse, 2024-04-08:
If you want to serve other html files you have to edit the code. I see no point in a fs browser to edit the HTML files directly when you can just use WebDav in Windows and Android (I know of). This is not a issue or feature request. No need for chatgpt reply that is out of date and wrong. What is you knowledge in arduino coding?

ldijkman, 2024-04-08:
i can do a bit of coding mostly i use  arduino ide 1 on linux mint 
in your code it is not direct clear to me
how to change server on / to send any file
also looks like file opening / closing is broken after wrong urls entered in adress bar

i do not use windows 
for me it would be handy to edit the webcam html directly on sd 
- and see results direct
i use ace editor for that and changed it a bit 
maybe have to find an sd version of the ace editor

ace editor on littlefs

[image]

ldijkman, 2024-04-08:
ill try to explain what i want

this is http://granden.local/bulbs.html
garden. local scans mdns for all devices in local network
and lists bulb.html of then iframed in bulbs.html

[image]

so garden.local bulbs.html has found http://esp32cam.local
but it does not have a http://esp32cam.local/bulb.html
so display in iframe is not found
i want http://esp32cam.local/bulb.html to serve an video stream of wificam
and on top a link to the main page at http://esp32cam.local

bit of automatic home assistant card overview with no programming
automatic mdns scan link list

s60sc, 2024-04-08:
to get a video feed in the iframe call ip_address/sustain?video=1

ldijkman, 2024-04-08:
yep figured out with chrome inspect
http://esp32cam.local/sustain?stream=0

could /data be in spiffs ?
would be easier for an inbrowser easy web flash to program the esp32-cam
to make it easier to flash/program for normal people

[image]

ldijkman, 2024-04-08:
> video=1

[image]

ldijkman, 2024-04-08:
utils.cpp

i do not know but i think this does not make sense to me
are you using ip as mdns name

mdns is for using human friendly names not ungueasable ip numbers
but 
http://frontview.local 
http://backyard.local

```
static void setupMdnsHost() {  
  // set up MDNS service 
  char mdnsName[MAX_IP_LEN]; // max mdns host name length
  snprintf(mdnsName, MAX_IP_LEN, hostName);
  if (MDNS.begin(mdnsName)) {
    // Add service to MDNS
    MDNS.addService("http", "tcp", HTTP_PORT);
    MDNS.addService("https", "tcp", HTTPS_PORT);
    //MDNS.addService("ws", "udp", 83);
    //MDNS.addService("ftp", "tcp", 21);    
    LOG_INF("mDNS service: http://%s.local", mdnsName);
  } else LOG_WRN("mDNS host: %s Failed", mdnsName);
  debugMemory("setupMdnsHost");
}

```

rjsachse, 2024-04-08:
Nothing wrong with mdns. You need to edit webserver.cpp.
If want you want the stream to output to bulb.hml you need to edit the code. Maybe create another xtask. Maybe this program is too advance for you yet, to edit to your liking.

Spend some time reading the code and understand the layout

closed 2024-04-09 by ldijkman