s60sc/ESP32-CAM_MJPEG2SD #407: inbrowser flash ESP32-CAM

ldijkman, 2024-04-10:
a test

inbrowser flash ESP32-CAM

https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

[image]

ldijkman, 2024-04-11:
https://youtu.be/YLLGBM3i2aQ

[image]

https://youtu.be/YLLGBM3i2aQ

s60sc, 2024-04-11:
do you want me to add a link to the video from the readme?

ldijkman, 2024-04-11:
> do you want me to add a link to the video from the readme?

its ok, no problem

ldijkman, 2024-04-11:
Thanks for the link
added backlink on bottom of page
https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

maybe note that it is only for the ESP32-CAM with USB port shield  ESP32-CAM-MB

ldijkman, 2024-04-12:
hello

maybe better on button reload data 
to have a message popup to confirm
to replace /data htm and js for github versions
now nothing is said to user

hostname might be changed so also mdns

[image]

ldijkman, 2024-04-13:
No SD-Card Version
Just Stream, No Record
webserver files in LittleFS

Easy Flash Install from Browser

https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

[image]

ldijkman, 2024-04-13:
<div class="flex gap-2 py-2.5 flex-row"><div class="text-xs text-[#6b7684] text-right"><div class="relative flex items-center justify-center"><svg xmlns="http://www.w3.org/2000/svg" viewBox="0 0 24 24" width="32" height="32"><g fill="none" fill-rule="evenodd"><path fill="#6436E2" d="M12 0c2.309 0 4.412.564 6.19 1.572a11.208 11.208 0 0 1 4.238 4.238C23.436 7.588 24 9.691 24 12c0 2.309-.564 4.412-1.572 6.19a11.208 11.208 0 0 1-4.238 4.238C16.412 23.436 14.309 24 12 24c-2.309 0-4.412-.564-6.19-1.572a11.208 11.208 0 0 1-4.238-4.238C.564 16.412 0 14.309 0 12c0-2.309.564-4.412 1.572-6.19A11.208 11.208 0 0 1 5.81 1.572C7.588.564 9.691 0 12 0z"></path><path fill="#FFF" d="M18.233 13.357a2.678 2.678 0 0 1-.55 1.406 2.733 2.733 0 0 1-1.23.896v-3.411a.46.46 0 0 0-.237-.408l-3.548-2.021 1.228-.7a.044.044 0 0 1 .041-.004l2.938 1.674c.45.257.817.634 1.058 1.09.24.454.344.967.3 1.478zm-2.735 2.203c-.001.715-.29 1.4-.802 1.906a2.757 2.757 0 0 1-3.682.168 2.14 2.14 0 0 0 .087-.048l2.906-1.656a.47.47 0 0 0 .238-.408V11.48l1.229.7a.044.044 0 0 1 .016.013.045.045 0 0 1 .008.02v3.347zM8.55 17.038a2.733 2.733 0 0 1-1.661-1.256 2.659 2.659 0 0 1-.326-1.808l.086.05 2.906 1.657a.478.478 0 0 0 .477 0l3.548-2.021v1.4a.043.043 0 0 1-.018.036l-2.937 1.674a2.77 2.77 0 0 1-2.075.268zM5.853 11.57a2.67 2.67 0 0 1 .272-2.047A2.716 2.716 0 0 1 7.548 8.34l-.001.099v3.312a.461.461 0 0 0 .238.408l3.548 2.02-1.228.7a.045.045 0 0 1-.042.004l-2.938-1.675a2.706 2.706 0 0 1-1.272-1.638zm2.648-3.131c0-.513.148-1.015.427-1.448a2.724 2.724 0 0 1 1.148-.995 2.769 2.769 0 0 1 2.91.37 2.18 2.18 0 0 0-.086.049L9.994 8.07a.47.47 0 0 0-.239.407l-.002 4.041-1.228-.7a.043.043 0 0 1-.024-.033V8.44zm3.5 1.76 1.58.9v1.8l-1.58.9-1.58-.9v-1.8l1.58-.9zm2.859-3.327a2.76 2.76 0 0 1 1.447.482c.426.293.759.7.959 1.173.2.473.26.992.172 1.498a3.618 3.618 0 0 0-.086-.051l-2.906-1.656a.478.478 0 0 0-.477 0l-3.548 2.02V8.94c0-.007 0-.014.004-.02a.043.043 0 0 1 .014-.017l2.937-1.672c.45-.256.965-.38 1.484-.358zm4.32 5.875a3.607 3.607 0 0 0-.93-2.054 3.543 3.543 0 0 0-.312-2.946 3.66 3.66 0 0 0-1.692-1.508 3.723 3.723 0 0 0-2.268-.233 3.633 3.633 0 0 0-1.24-.895 3.679 3.679 0 0 0-1.502-.31 3.712 3.712 0 0 0-2.17.691 3.631 3.631 0 0 0-1.338 1.82c-.507.103-.987.31-1.406.611-.42.3-.769.685-1.025 1.13a3.589 3.589 0 0 0 .452 4.254 3.544 3.544 0 0 0 .313 2.946 3.66 3.66 0 0 0 1.692 1.507 3.723 3.723 0 0 0 2.268.234c.344.382.766.687 1.239.895a3.68 3.68 0 0 0 1.503.31c.78.003 1.54-.24 2.17-.691a3.63 3.63 0 0 0 1.339-1.822c.507-.102.987-.31 1.406-.61.42-.3.769-.686 1.025-1.13.39-.665.557-1.435.476-2.2z"></path></g></svg></div></div><div class="max-w-[85%] px-3 break-words text-[#191f28] leading-6 rounded-xl py-2.5 mr-auto rounded-tl-md bg-white"><div class="prose max-w-[100%] break-words"><p>To add a confirmation dialog when the "deldata" button is clicked, you can modify the <code>&lt;svg&gt;</code> element associated with the "deldata" action and include an <code>onclick</code> event that triggers a confirmation dialog using JavaScript's <code>confirm()</code> function. If the user confirms the action, the intended function to reload or clear the data can be executed.</p>
<p>Below, I've added an <code>onclick</code> attribute with a confirmation dialog to the "deldata" button in the provided HTML code. This example assumes you have a JavaScript function or logic to handle the "deldata" action, which should be called if the user confirms the action.</p>
<p>Find the section for the "deldata" button and modify it as follows:</p>
<pre class="scrollbar-thin scrollbar-thumb-[#33435a] scrollbar-track-[#1f2937]"><div style="color: rgb(248, 248, 242); background: rgb(43, 43, 43); font-family: Consolas, Monaco, &quot;Andale Mono&quot;, &quot;Ubuntu Mono&quot;, monospace; text-align: left; white-space: pre; word-spacing: normal; word-break: normal; overflow-wrap: normal; line-height: 1.5; tab-size: 4; hyphens: none; padding: 1em; margin: 0.5em 0px; overflow: auto; border-radius: 0.3em;"><code class="language-html" style="color: rgb(248, 248, 242); background: none; font-family: Consolas, Monaco, &quot;Andale Mono&quot;, &quot;Ubuntu Mono&quot;, monospace; text-align: left; white-space: pre; word-spacing: normal; word-break: normal; overflow-wrap: normal; line-height: 1.5; tab-size: 4; hyphens: none;"><span class="token" style="color: rgb(254, 254, 254);">&lt;</span><span class="token" style="color: rgb(255, 160, 122);">div</span><span class="token" style="color: rgb(254, 254, 254);">&gt;</span><span>
</span><span>  </span><span class="token" style="color: rgb(254, 254, 254);">&lt;</span><span class="token" style="color: rgb(255, 160, 122);">svg</span><span class="token" style="color: rgb(255, 160, 122);"> </span><span class="token special-attr" style="color: rgb(171, 227, 56);">onclick</span><span class="token special-attr attr-equals" style="color: rgb(254, 254, 254);">=</span><span class="token special-attr" style="color: rgb(254, 254, 254);">"</span><span class="token special-attr value javascript language-javascript" style="color: rgb(255, 215, 0);">confirmDelDataAction</span><span class="token special-attr value javascript language-javascript" style="color: rgb(255, 215, 0);">(</span><span class="token special-attr value javascript language-javascript" style="color: rgb(255, 215, 0);">)</span><span class="token special-attr" style="color: rgb(254, 254, 254);">"</span><span class="token" style="color: rgb(254, 254, 254);">&gt;</span><span>
[... cut ...] uld replace the placeholder code with your actual implementation for handling the "deldata" action.</p>
<p>Remember, the <code>confirmDelDataAction()</code> function is a custom function that you need to define based on the specific requirements of your application. The <code>sendControl()</code> function mentioned in the comment is just a placeholder and should be replaced with the actual function call that triggers the "deldata" action on your ESP32-CAM or related system.</p></div></div></div>

ldijkman, 2024-04-16:
hello
if i understand it right from code

reload /data reloads overwrtite webserver files on sd or littlefs
clear nvs clears dat from eeprom wifipassword?

wouldnt it be better to give user more info and cancel/confirm message

[image]

s60sc, 2024-04-16:
Your understanding is correct.
I dont want to 'baby' users, they should read the code.

ldijkman, 2024-04-16:
> Your understanding is correct. I dont want to 'baby' users, they should read the code.

well some people wil not be interested in coding 
but wile like your application

added a title to the divs for some popup info

https://github.com/ldijkman/async-esp-fs-webserver/blob/0efc2565e9b5a782ea9f81424c722a048dd08a13/docs/ino/ESP32-CAM_Camera_Webserver/BIN/data/MJPEG2SD.htm#L1569

https://github.com/ldijkman/async-esp-fs-webserver/blob/0efc2565e9b5a782ea9f81424c722a048dd08a13/docs/ino/ESP32-CAM_Camera_Webserver/BIN/data/MJPEG2SD.htm#L1575

[image]

[image]

sjude68, 2024-04-18:
> Your understanding is correct. I dont want to 'baby' users, they should read the code.

I didn't know that the data files are taken from github /data

If this is the case then there is no need to copy the /data files to the SDcard when I reflash the new version.  Is that correct ?

s60sc, 2024-04-18:
if you have an internet connection they will be automatically downloaded

ldijkman, 2024-04-19:
> > Your understanding is correct. I dont want to 'baby' users, they should read the code.
> 
> I didn't know that the data files are taken from github /data
> 
> If this is the case then there is no need to copy the /data files to the SDcard when I reflash the new version. Is that correct ?

you get a simpler wifi router config connect page when /data is not present

and files are downloaded

---

and on button reload /data  /data is deleted
esp is rebooted 
and start downloading from github

web serial monitor output with clickable links
```

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

[03:42:57.588 logSetup] Setup RAM based log, size 7168, starting from 7154

[03:42:57.589 logSetup] =============== ESP-CAM_MJPEG 9.7 ===============
[03:42:57.599 WARN initBrownout] Brownout warning previously notified 
[03:42:57.599 logSetup] Compiled with arduino-esp32 v2.0.14
[03:42:57.610 printResetReason] Software reset via esp_restart
[03:42:57.610 printWakeupReason] Wakeup by reset
[03:42:57.723 infoSD] SD card type SDHC, Size: 7.4GB
[03:42:57.723 listFolder] Sketch size 1.6MB
[03:42:58.031 listFolder] SD_MMC: 505.4MB used of 7.4GB
[03:42:58.032 loadConfig] Load config
[03:42:58.105 checkConfigFile] Created /data/configs.txt from local store
[03:42:58.236 updateAppStatus] Disabling motion detection
[03:42:58.250 setup] PSRAM size: 8.0MB
[03:42:58.447 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[03:42:58.719 setWifiSTA] Wifi Station IP from DHCP
.[03:42:58.835 onWiFiEvent] 
WiFi Station connection to Bangert_30_Andijk, using hostname: ESP-CAM_MJPEG_B8472CB267AC
......[03:43:01.857 onWiFiEvent] 
Wifi Station IP, use 'http://10.10.100.101' to connect
[03:43:02.258 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_B.local
[03:43:04.528 startWifi] Wifi stats for Bangert_30_Andijk - signal strength: -61 dBm; Encryption: WPA2_PSK; channel: 11
[03:43:04.530 startPing] Started ping monitoring - Off
[03:43:04.544 startWebServer] Starting web server on port: 80
[03:43:04.545 startWebServer] Remote server certificates not checked
[03:43:04.545 startSustainTasks] Started 1 HTTP sustain tasks
[03:43:04.555 prepUpload] File uploads will use FTP server
[03:43:04.556 prepTelegram] Telegram not being used
[03:43:04.556 resetWatchDog] WatchDog started using task: ping
[03:43:04.567 getLocalNTP] Using NTP server: pool.ntp.org
[03:43:04.579 showLocalTime] Got current time from NTP: 19/04/2024 03:43:04 with tz: GMT0
[03:43:04.609 updateAppStatus] Disabling motion detection
[03:43:04.794 prepRecording] To record new AVI, do one of:
[03:43:04.794 prepRecording] - press Start Recording on web page
 
[03:43:04.794 prepRecording] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[03:43:04.805 checkMemory]  Free: heap 49000, block: 23540, min: 46684, pSRAM 1490603
[03:43:04.816 loop] =============== Total tasks: 17 ===============

[03:43:05.376 wgetFile] Downloading /data/common.js from /ldijkman/async-esp-fs-webserver/master/docs/ino/ESP32-CAM_Camera_Webserver/BIN/data/common.js
[... 55 lines cut ...]
[03:54:10.309 resetWatchDog] WatchDog started using task: ping
[03:54:10.315 getLocalNTP] Using NTP server: pool.ntp.org
[03:54:10.318 startWebServer] Starting web server on port: 80
[03:54:10.327 startWebServer] Remote server certificates not checked
[03:54:10.337 showLocalTime] Got current time from NTP: 19/04/2024 03:54:10 with tz: GMT0
[03:54:10.337 startSustainTasks] Started 1 HTTP sustain tasks
[03:54:10.348 prepUpload] File uploads will use FTP server
[03:54:10.349 prepTelegram] Telegram not being used
[03:54:10.399 updateAppStatus] Disabling motion detection
[03:54:10.583 prepRecording] To record new AVI, do one of:
[03:54:10.583 prepRecording] - press Start Recording on web page
 
[03:54:10.585 prepRecording] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[03:54:10.596 checkMemory]  Free: heap 49064, block: 23540, min: 46748, pSRAM 1490603
[03:54:10.608 loop] =============== Total tasks: 17 ===============

[03:54:11.143 wgetFile] Downloading /data/common.js from /ldijkman/async-esp-fs-webserver/master/docs/ino/ESP32-CAM_Camera_Webserver/BIN/data/common.js
[03:54:11.669 wgetFile] Downloaded /data/common.js, size 28604 bytes
[03:54:12.392 wgetFile] Downloading /data/MJPEG2SD.htm from /ldijkman/async-esp-fs-webserver/master/docs/ino/ESP32-CAM_Camera_Webserver/BIN/data/MJPEG2SD.htm
[03:54:14.004 wgetFile] Downloaded /data/MJPEG2SD.htm, size 95KB
[03:54:26.926 wsHandler] Websocket connection: 54
[03:54:40.300 setAlarm] Alarm scheduled at 20/04/2024 01:00:00
[03:54:41.329 saveConfigVect] Config file saved 
[03:54:41.355 doAppPing] Daily rollover
```

ldijkman, 2024-04-20:
color in web serial terminal alike log color codes
find ip mdns in a blink

[image]

ldijkman, 2024-04-20:
reload /data
added a button test with confirm popup

https://github.com/ldijkman/async-esp-fs-webserver/blob/master/docs/ino/ESP32-CAM_Camera_Webserver/BIN/data/MJPEG2SD.htm#L1590

[image]

ldijkman, 2024-04-21:
aded an install button for another board
but should be faster but i find it slow
must be my arduino coding / settings

# ESP32-S3-Wroom-1, 16Mb?, OV2640 Camera, SD-Card Slot, Dual USB 16Mb?

https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

[image]

closed 2024-04-30 by ldijkman

guece1, 2024-04-30:
@ldijkman

Nice, thank you for supporting this Board 

i run into a problem here, i have an OV560 Cam attached it won´t work
Could you please implement an option to choose the right Cam
also i think there is a need to change the FB_BUFFERS size for the OV5640

@[s60sc](https://github.com/s60sc) commented [Apr 18, 2024](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/411#issuecomment-2063761346)

Bug awaiting fix due to combination of S3 and OV5460 trying to allocate too much PSRAM

In the interim, in appGlobals.h, change:
#define FB_BUFFERS 12 // 1 being processed, rest being filled
to:
#define FB_BUFFERS 6 // 1 being processed, rest being filled

my log:

[00:00:00.018 logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.018 logSetup] =============== ESP-CAM_MJPEG 9.7 ===============
[00:00:00.029 logSetup] Compiled with arduino-esp32 v2.0.14
[00:00:00.029 printResetReason] Power on reset
[00:00:00.039 printWakeupReason] Wakeup by reset
[00:00:00.102 infoSD] SD card type SDHC, Size: 59.5GB
[00:00:00.103 listFolder] Sketch size 1.5MB
[00:00:00.239 listFolder] File: /data/configs.txt, size: 5707 bytes
[00:00:00.244 listFolder] File: /data/common.js, size: 34155 bytes
[00:00:00.248 listFolder] File: /data/MJPEG2SD.htm, size: 90KB
[00:00:00.249 listFolder] SD_MMC: 3.9MB used of 59.0GB
[00:00:00.259 loadConfig] Load config
[00:00:00.456 updateAppStatus] Disabling motion detection
[00:00:00.459 setup] PSRAM size: 8.0MB
[00:00:02.438 setWifiSTA] Wifi Station IP from DHCP
[00:00:02.451 WARN startWifi] SSID  not available, use AP 
[00:00:02.453 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_D8AAEFE22748
[00:00:02.453 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_D8AAEFE22748 started, use 'http://192.168.4.1' to connect
[00:00:02.475 setupMdnsHost]  mDNS service: http://ESP-CAM_MJPEG_D.local 
[00:00:05.685 startPing] Started ping monitoring - Off
[00:00:05.687 startWebServer] Starting web server on port: 80
[00:00:05.688 startWebServer] Remote server certificates not checked
[00:00:05.698 WARN setup] Startup Failure: Camera init error 0xffffffff on CAMERA_MODEL_ESP32S3_EYE_FREENOVE 
[00:00:05.708 loop] =============== Total tasks: 13 ===============

[00:00:10.686 resetWatchDog] WatchDog started using task: ping

Thank you in advance