s60sc/ESP32-CAM_MJPEG2SD #409: [Feature Request] Remove SD card requirement.

Sokoloft, 2024-04-11:
As title says. It'd be nice to remove the necessity to have a SD card. So I can just stream it over my network to ispy. 

If there is a way to do this already I must've missed the configuration.

rjsachse, 2024-04-11:
Haven't tried myself yet, but I think you can change this
https://github.com/s60sc/ESP32-CAM_MJPEG2SD/blob/2a609256d434f85447ec6700f6c2d92aeb190a42/appGlobals.h#L109 to SPIFFS
Haven't tried myself yet. Make sure you have right partition selected too

Sokoloft, 2024-04-11:
Cool. Well that does seem to work? However trying to access the web page. The page displays "The URI does not exist". Is there a way to toggle the MJPEG stream? From the code? I keep getting the following in the serial monitor.

```
[12:47:09.353 WARN remoteServerConnect] Timed out connecting to server: raw.githubusercontent.com, Err: RSA - The public key operation failed : BIGNUM - Memory allocation failed 
[12:47:15.848 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm 
```

s60sc, 2024-04-11:
Need to compile with this [file](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/blob/master/extras/partitions.csv) in sketch folder

Sokoloft, 2024-04-11:
Sweet. Got it working. Thank you.

ldijkman, 2024-04-12:
Hello
Would like to have webserver files in spiffs or littlefs
So that i can flash them in browser install
So that it works without having to create an sd card with webserverfiles

But still have the sdcard used for recording 

Greet luberth

ldijkman, 2024-04-13:
added an orange  button no sd card flash from browser install

https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

[image]