s60sc/ESP32-CAM_MJPEG2SD #410: https setup?

TinkerType, 2024-04-11:
Hi s60sc and everyone who tried this... I am trying to set up https for the web server, so far without success. http is working alright, but I would really prefer https with a self-signed certificate.

I followed the comments in certificates.cpp and entered the command on line 21 with my target IP to create the prvtkey_pem and cacert_pem files. So far so good....

[image]

I replaced the lines for the private key and certificate with the respective lines from the files.

[image]

[image]

Saved it, compiled and uploaded it to the board. 

Then on the access point, I configured 'Access Settings' to 'Use HTTPS' and under 'Edit Config' -> 'Wifi' I entered the target IP in the field 'Static IP address'.

[image]

[image]

Alright... Saved the config and then 'Reboot ESP'. 

At this point Chrome already refused to serve the page, which was expected, so I entered the certificates under 'Security' -> 'Manage Certificates' -> 'Trusted Freaking Root Certification Authorities'. The wizard allowed browsing to the 'cacert.pem' file, which I selected, Confirmed all buttons.

[image]

[image]

Target IP shows up in the list, all seems good... but no success. Browsing to the page still shows 'This site can't be reached'. 

[image]

What is interesting is that trying http://<target ip> redirects to https://<target ip>, so something must be there. 

The log from the serial monitor says, there is a 'Wifi Station IP' using https, so that is promising, plus the web server is listening on 443, which also seems like a good start. 

[image]

Btw, I did check the firewall in my router and previously with http;//<target ip> all was working fine.

If anyone knows, what's the missing piece or what steps to take from here, I would greatly appreciate it. Thanks very much.

rjsachse, 2024-04-11:
I don't know if you need to check (Check Certs) under use https

s60sc, 2024-04-11:
that static ip looks odd, does it work with http?

check certs is for external sites confidence (middle man attack)

you have to get certs exactly right, a tiny mistake will fail

only use https on ESP32S3, ESP32 has insufficient memory

TinkerType, 2024-04-11:
Thanks very much rjsachse and s60sc. I just realized my whole idea was not going to work anyway, i.e. to use https through port forwarding with just an IP. Perhaps I will try a reverse proxy instead. I appreciate your responses!