s60sc/ESP32-CAM_MJPEG2SD #411: Camera init error 0xffffffff with esp32s3 and OV5640

guece1, 2024-04-13:
I have this problem with my esp32s3 and OV5640 (no problem with OV2640)

my esp32s3 is this https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board

Compilation error: 'esp_spiram_get_size' was not declared in this scope

this problem exist on Arduino core for the ESP32 3.0.0-rc1 and 3.0.0-alpha3

on 2.0.15 and 2.0.14 i can compile but my cam module wont startup
Startup Failure: Camera init error 0xffffffff on CAMERA_MODEL_ESP32S3_EYE_FREENOVE

my configs 
Board = ESP32S3 Dev Module
flash size 8mb
8m with spiffs
tried opi Psram and qspi Psram and no Psram

s60sc, 2024-04-18:
Bug awaiting fix due to combination of S3 and OV5460 trying to allocate too much PSRAM

In the interim, in `appGlobals.h,` change:
`#define FB_BUFFERS 12 // 1 being processed, rest being filled`
to:
`#define FB_BUFFERS 6 // 1 being processed, rest being filled`

guece1, 2024-04-18:
appreciate for your time and help.  now everything works

s60sc, 2024-05-04:
should be fixed in v9.7.1

closed 2024-05-04 by s60sc