s60sc/ESP32-CAM_MJPEG2SD #416: [question] Migration from 2.x to 3.0

guece1, 2024-04-17:
I there in near future a plan for the migration of the project from versions 2.X (based on ESP-IDF 4.4) to version 3.0 (based on ESP-IDF 5.1) of the Arduino ESP32 core. ?
I read that the Support for OV5640 was dropped in actual Arduino framework. But in 3.0.1RC it is back in

s60sc, 2024-04-17:
Not until stable release

guece1, 2024-04-17:
did i missing something 
Is there anothere way to get the OV5640 working on esp32s3  
this is my board https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board
i try for weeks now reading codes without succes 
no baby mode here

s60sc, 2024-04-17:
Unlike ov2640, some ov5640 have a non standard pinout. Check the wiring spec

guece1, 2024-04-18:
I tried for weeks ,the wiring specs are are the same it works with an esp32 cam.

My esp32s3 works with the OV5640 on other Sketches, like the newest Tasmota  compiled with Arduino ESP32 Core 3.0.1RC.

Can anyone confirm that the OV5640 dont work on esp32-s3 but on esp32 on the actual Arduino framework. ?

s60sc, 2024-04-18:
Tried an OV5640 on my freenove board and got same problem. Could by an issue with later arduino 2.X releases, but I'll investigate furrther as time allows

guece1, 2024-04-18:
thank you for your time

I read from the Tasmota git

Support is removed in IDF and in Arduino precompiled libs 2.0.14 to save code space.

Modules OV3660 and OV5640 will be supported in next Arduino 3.0.0 alpha Platform build. Modules will NOT be added to core 2.0.14

On Tasmota 13.4.1.1 compiled on Core/SDK Version | 3_0_0/5.1.3.240325 the OV5640 works again

Could this be the problem ? 
Can we compile 2.0.14 libs ourself to implement it back ? 
Or is the only way to wait for the stable Arduino 3.0.0 Platform build ?

s60sc, 2024-04-18:
Sorry it is my fault. The combination of S3 and OV5460 was trying to allocate too much PSRAM

In the interim, in `appGlobals.h,` change:
`#define FB_BUFFERS 12 // 1 being processed, rest being filled`
to:
`#define FB_BUFFERS 6 // 1 being processed, rest being filled`

guece1, 2024-04-18:
appreciate for your time and help it works very well now on core 2.0.15

s60sc, 2024-05-04:
should be fixed in v9.7.1

closed 2024-05-04 by s60sc

rjsachse, 2024-06-04:
Not sure if you know that version 3 is out not as stable

s60sc, 2024-06-04:
working on conversion but will take time due to api changes