s60sc/ESP32-CAM_MJPEG2SD #418: re: CAMERA_MODEL_NEW_ESPS3_RE1_0 SD_MMC pins

OldYin, 2024-04-20:
I know you intended to get these right but a minor typo makes them wrong. I know that anybody who actually owns one of these boards, won't be confused... for long but I figured you would like to correct this.

In camera_pins.h
You have:

//Define SD Pins
#define SD_MMC_CLK 42
#define SD_MMC_CMD 39
#define SD_MMC_D0 4

Correct pins are:

//Define SD Pins
#define SD_MMC_CLK 42
#define SD_MMC_CMD 39
#define SD_MMC_D0 41

Regards,
Alex

s60sc, 2024-04-20:
Thanks. I dont have this board myself so see if @rjsachse will confirm as they provided the update

rjsachse, 2024-04-20:
Good morning, I don't have that board either. I just copied the pins from appGlobals.h and pasted in camera_pins.h I noticed the board got added in march in version 9.62. Surprisingly you are the first to noticed. Easy fix.

rjsachse, 2024-04-20:
Pull Request done

OldYin, 2024-04-21:
If I may refer you to issue #379 in which mamazad7 posted full pin details (with diagrams) for this board. Hope that helps. 
Also I DO have this board :)

regards,
Alex

rjsachse, 2024-04-21:
Been fixed just waiting on @s60sc to accept pull request, it was a typo in version 9.62.

OldYin, 2024-04-21:
No worries; thanks :)

commit 7925d0a5 (2024-04-21): Merge pull request #419 from rjsachse/patch-1

fix #418
closed 2024-04-21 by s60sc