s60sc/ESP32-CAM_MJPEG2SD #421: Hashtag in SSID wont connect

adix992, 2024-04-26:
So I have never had this problem before and I run a Home Assistant with 20+ integrations and 60+ devices just for reference.

I have a SSID which has a #1234 in the name which I use for smart devices. But after hours of trying to connect a ESP32S3 with this project, I realized it keept rejecting my WiFi setup as it tries to connect to a SSID without the #1234.

Log :
[00:00:06.451 WARN startWifi] SSID Smart Devices  not available, use AP 

Config :
char ST_SSID[MAX_HOST_LEN]  = "Smart Devices #1234"; //Default router ssid
char ST_Pass[MAX_PWD_LEN] = "2540254025"; //Default router passd

Same error if I try to enter the SSID from the webui from the local AP. I hope this can be remedied somehow, would be a whole day task to change SSID of all smart devices just to add ESPCAM with this project which I otherwise love (And I might or might not have 4 of these sucker lying around now :/)

s60sc, 2024-04-26:
The problem is that # is a special URI char.
As an interim in `prefs.cpp`, try changing:
    `else if (!strcmp(variable, "ST_SSID")) strncpy(ST_SSID, value, MAX_HOST_LEN-1);`
with:
    `else if (!strcmp(variable, "ST_SSID")) strncpy(ST_SSID, "Smart Devices #1234", MAX_HOST_LEN-1);`

Not tested though as dont have access to an esp32 right now

rjsachse, 2024-04-26:
Also do have allow white spaces in config

adix992, 2024-04-27:
Thanks for the input,

After some tinkering I found a workaround :
Used old phone to setup Hotspot AP with no hashtag in name
Connected the ESP to the Hotspot
From Hotspot phone I could access the webUI
From here I could change the WiFi SSID settings to Smart Devices #1234

After restart it connected to the Smart Devices #1234 AP

Slightly longer onboard proces, but not to painful and no code tinkering needed

s60sc, 2024-05-04:
should be fixed in v9.7.1

closed 2024-05-04 by s60sc