s60sc/ESP32-CAM_MJPEG2SD #422: Remote access to the stream

matin-alizadeh, 2024-04-28:
Hi.
I want to access the stream through the Internet remotely.
I cannot use the port forwarding solution.
Is there another way to access the stream over the internet?

rjsachse, 2024-04-30:
Can you access your home network through the internet? Have you tried reverse proxy

matin-alizadeh, 2024-05-01:
> Can you access your home network through the internet? Have you tried reverse proxy

No It's not possible to access my home network via the Internet.
I want the ESP to send its stream to a server and I can view the stream through that server.

rjsachse, 2024-05-01:
Do you have a server set up? You could push the mjpeg stream. If you can't access port forwarding ask your ISP to turn off CGNAT. 

What is the server setup to be able to pull the images?

matin-alizadeh, 2024-05-01:
I want to use a virtual server for this.
I do not intend to use port forwarding and I am looking for an alternative solution.

rjsachse, 2024-05-01:
Ok, I have tried to help you out, your not giving any details of the server or capabilities, is this just an idea? You could push the images to a server or a server could pull the images.  Maybe try blynk.

closed 2024-05-04 by s60sc