s60sc/ESP32-CAM_MJPEG2SD #426: SW_RESET ISSUE

wildmack, 2024-05-04:
Hello, I have no idea what I did to mess this up.  I had it up and running on one of my ESP32 cam boards, then I went to load it on another board and now every board I try I get the below message.  I tried 5V power supply and USB power and no difference.  I must have changed something, but I have no idea what it was. Any idea what the heck I did??  This is driving me nuts !!  Thanks Dave.....

20:26:03.440 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:26:03.440 -> configsip: 0, SPIWP:0xee
20:26:03.440 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:26:03.440 -> mode:DIO, clock div:1
20:26:03.440 -> load:0x3fff0030,len:1344
20:26:03.472 -> load:0x40078000,len:13964
20:26:03.472 -> load:0x40080400,len:3600
20:26:03.472 -> entry 0x400805f0
20:26:03.472 -> ets Jun  8 2016 00:22:57

s60sc, 2024-05-04:
bootloop, maybe indequate power or pin conflict compiled with wrong board selected

wildmack, 2024-05-04:
I tried an external power supply and still the same issue.  I have nothing connected to the pins, I'm just trying to run the basic code.  Can you please take a look at the Pic's and see if anything looks out of place.  Thanks Dave......

[image]
[image]
[image]
[image]
[image]

guece1, 2024-05-04:
Check in the Boards Manager the Version from "Esp32 by Espressif Systems" it should be not higher than 2.0.15 or better 2.0.14

[image]

ldijkman, 2024-05-05:
maybe try to flash it from browser
https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

then you krow if it is software or hardware related

wildmack, 2024-05-07:
Hello, and thank you.  
I flashed from browser and that got me closer.  It first took me to a screen that I entered my SSID and password, then in the serial monitor it gave me the IP address the camera was on.  But when I open that page the only button that works is the Stream button and I can see the stream from the camera.  But, I can't open any other button, so I can't configure the camera.  Any suggestions??
[image]

ldijkman, 2024-05-07:
Hmmm lefttop button camera 
should not say text camera
But your text camera type ov2640
Sorry i have no idea

wildmack, 2024-05-08:
It was user error.  I didn't put the two files into the data folder.  But after doing that it seems to work fine.  I still can't load using the Arduino IDE, any ideas what could be causing that?  Also I want to use the 5640 camera, do I need to change anything to do that?   Thanks Dave.....

ldijkman, 2024-05-08:
If there is a wifi router internet connection 
files should be auto downloaded and saved to sd card

wildmack, 2024-05-09:
Yes, I tried that and the files downloaded.  I tried a reinstall and now nothing works again?? I'm lost...  Is this an error message?
[image]

davemaster, 2024-05-09:
> Hello, I have no idea what I did to mess this up. I had it up and running on one of my ESP32 cam boards, then I went to load it on another board and now every board I try I get the below message. I [...]

Greetings,

I think is a power issue, the best way to START, is plug the camera IN A POWER SUFFICIENT USB port, check (manual, datasheet, etc) if your laptop, computer, etc is able to give the current to run ESP32-CAM and the ESP32-CAM-MAINBOARD

rjsachse, 2024-05-09:
Do you have the right partition selected

wildmack, 2024-05-12:
Thank you all for the help.  I finally got it working, I changed my SD card and everything loaded completely.  Thanks you again.  Dave....

closed 2024-05-12 by wildmack

davemaster, 2024-05-13:
Great, I You can, share a video running... best regards