s60sc/ESP32-CAM_MJPEG2SD #427: Telegram message notification

ldijkman, 2024-05-05:
Telegram message notification

would it not be nice to have a link to the sttream in the telegram message?

not so nice wide angle night vision cam

[image]

s60sc, 2024-05-05:
I dont think the telegram bot api supports streaming, its message based.

ldijkman, 2024-05-06:
i was think more like a doorbell cam
if input is high or low send telegram message
with a handy stream link that will open in browser 
a message like next

doorbell  
see http://localip/stream  http://externalip/stream

s60sc, 2024-05-06:
not sure what value is, you could bookmark url

closed 2024-06-01 by s60sc