s60sc/ESP32-CAM_MJPEG2SD #431: Card in Home Assistant

BeHappyYou, 2024-05-12:
Hello,

Is it possible to use a (picture) card to have a stream of the ESP32 camera visible in Home Assistant.

When I use a picture card with only the IP Address it doesn't show me anything.
When I use a website card it shows me the menu (as expected :-))

Thanks!

s60sc, 2024-05-15:
[port forward](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=7%3Bfps%3D1-,Port%20Forwarding,-To%20access%20the)

BeHappyYou, 2024-05-18:
@s60sc Thank you!

I did look further into it to see if it is possible to prevent ‘port forward’.

I did the following:
1)	At the camera's webpage I activated ‘Edit Config -> Motion -> "Enable NVR Video stream: /sustain?video=1"’
2)	At the Home Assistant I added ‘Devices & Services -> Add integration -> Generic Camera: rtsp stream: http://IP_ADDRESS/sustain?video=1 and selected HTTP.’
3)	I added the "Frigate Lovelace Card" under HACS.
4)	In the Overview section I added the Frigate Lovelace Card and selected  the Generic Camera.

That’s it.

You don't need to install Frigate of Go2RTC or any other addidtional add-on.

Hope this helps and thanks a lot for the excellent work you are doing! 😊

closed 2024-05-26 by BeHappyYou