s60sc/ESP32-CAM_MJPEG2SD #432: ESP32S3 (SPI_FAST_FLASH_BOOT) Issue

wildmack, 2024-05-15:
I tried to upgrade from the ESP32 Cam to the ESP32S3 Cam to eliminate the lag issues I was having, but now in the serial monitor I get the below message message.  I have tried different USB cables but I still get the same message.  I have the board identified as ESP32S3 Dev Module and the following settings:
// User's ESP32 cam board
#if defined(CONFIG_IDF_TARGET_ESP32)
#define CAMERA_MODEL_AI_THINKER 

// User's ESP32S3 cam board
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
//#define CAMERA_MODEL_FREENOVE_ESP32S3_CAM
#define CAMERA_MODEL_XIAO_ESP32S3 
#endif

This is the error message:
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdada
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
ESP-ROM:esp32s3-20210327

Any Ideas?  Thanks Dave.......

bjesuslopezg, 2024-05-16:
You need to be sure that the ESP32S3 you're using is actually N8R8 (Flash size 8 MB and PSRAM 8 MB). Once you validated that, then you need to make the right configurations in the Arduino IDE

Open the project with Arduino IDE and go to "Tools" 

Select Flash Size: "8 MB (64 MB)" 
Select Partition scheme: "8M with spiffs (3 MB APP/1.5MB SPIFFS)"
Select PSRAM: " QSPI PSRAM"

rjsachse, 2024-05-17:
Do you have usb cdc enabled

bjesuslopezg, 2024-05-17:
> Do you have usb cdc enabled

No

wildmack, 2024-05-17:
Should usb cdc be enabled ?  Thanks Dave....

rjsachse, 2024-05-17:
If your board is CAMERA_MODEL_XIAO_ESP32S3 then yes cdc should be enabled

wildmack, 2024-05-18:
Thank you.  Got everything working.

closed 2024-05-18 by wildmack