s60sc/ESP32-CAM_MJPEG2SD #433: Record save time in forced record

alexandrumot, 2024-05-18:
Firstly, great library. I didn't expect this module to support so many features. 
All I want is to record indefinitely and to save a video to SD every 10 secs. I've looked at #271 but without success. What I tried was to change maxFrames to close the video earlier:
```
bool useMotion  = true; // whether to use camera for motion detection (with motionDetect.cpp)
bool dbgMotion  = false;
bool forceRecord = true; // Recording enabled by rec button

// motion detection parameters
int moveStartChecks = 5; // checks per second for start motion
int moveStopSecs = 2; // secs between each check for stop, also determines post motion time
int maxFrames = 100; // maximum number of frames in video before auto close 
```
But it doesn't affect forcedRecord at all.

Arduino IDE 2.3.2
AI Thinker ESP32-CAM with FTDI

Any help is much appreciated.

s60sc, 2024-05-18:
Thanks.

Not designed for that purpose, but could:

- change `WiFi connect timeout (secs)` to 10
- modify `doAppPing()` which is now called every 10 secs to set  `forceRecord `off and on

but will need to experiment

alexandrumot, 2024-05-18:
Ok, I understand. Will experiment with pirPin to keep it recording with gaps for saving video to SD.

closed 2024-05-18 by alexandrumot