s60sc/ESP32-CAM_MJPEG2SD #436: OV5640 not detected

GoostYTQC, 2024-05-20:
Startup Failure: Camera init error 0xffffffff on CAMERA_MODEL_AI_THINKER
i try whit the default example and it work correctly but with this code i just throw me a error

davemaster, 2024-05-20:
Turn the power off, and carefully, using lens, check if you plugged the camera correctly, check the flex... take a picture

GoostYTQC, 2024-05-20:
[image]

GoostYTQC, 2024-05-20:
i just try white [esp32-cam-webserver](https://github.com/easytarget/esp32-cam-webserver) and it work so something in the code does detect the camera correctly, or i need to change something in the code ?

davemaster, 2024-05-20:
What ESP32 based camera have You got? You have to change the code according the module you are going to use... picture (complete) of your camera module pls.

GoostYTQC, 2024-05-21:
[image]

GoostYTQC, 2024-05-21:
its from this https://www.amazon.ca/dp/B07WSCJ78V?psc=1&ref=ppx_yo2ov_dt_b_product_details

GoostYTQC, 2024-05-21:
`#include "esp_camera.h"
#include "Arduino.h"

// Define the camera configuration (you might need to adjust these settings for your specific board)
#define PWDN_GPIO_NUM 32
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 0
#define SIOD_GPIO_NUM 26
#define SIOC_GPIO_NUM 27
#define Y9_GPIO_NUM 35
#define Y8_GPIO_NUM 34
#define Y7_GPIO_NUM 39
#define Y6_GPIO_NUM 36
#define Y5_GPIO_NUM 21
#define Y4_GPIO_NUM 19
#define Y3_GPIO_NUM 18
#define Y2_GPIO_NUM 5
#define VSYNC_GPIO_NUM 25
#define HREF_GPIO_NUM 23
#define PCLK_GPIO_NUM 22

void setup() {
  Serial.begin(115200);
  Serial.println("Starting camera...");

  // Camera configuration
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  // Select the appropriate frame size
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  // Camera initialization
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x\n", err);
    return;
  }

  // Get the camera sensor
  sensor_t *s = esp_camera_sensor_get();
  if (s == NULL) {
    Serial.println("Camera sensor get failed");
    return;
  }

  // Print the sensor PID
  Serial.print("Sensor PID: 0x");
  Serial.println(s->id.PID, HEX);  // Print the PID in hexadecimal format
}

void loop() {
  // Add your main code here, to run repeatedly
}
`
i used this code to make sure it get the right id and it output Sensor PID: 0x5640

davemaster, 2024-05-21:
That camera module, is not the one included with the ESP32-CAM, have You confirm it is compatible? 

I just run the code, without problems, assure You change the camera to 

[image]

I used Arduino IDE to compile the code... I'm very busy now, but tomorrow I will test it in my own ESP32-CAM module

GoostYTQC, 2024-05-21:
> That camera module, is not the one included with the ESP32-CAM, have You confirm it is compatible?
> 
> I just run the code, without problems, assure You change the camera to
> 
> ![imagen](https://private-user-images.githubusercontent.com/1075807/332247259-1a669cd5-b7b3-4864-8208-f0ec0eb864ff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYyNTU1MjYsIm5iZiI6MTcxNjI1NTIyNiwicGF0aCI6Ii8xMDc1ODA3LzMzMjI0NzI1OS0xYTY2OWNkNS1iN2IzLTQ4NjQtODIwOC1mMGVjMGViODY0ZmYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjFUMDEzMzQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWE0ODZiMjc0ZjVhMzVkYWNlNjdhMWZkYmJlN2JhZjdmMzI2OTAwNjlhNDEyNzlhZmFkZTBlM2MyZjU0YmU4NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.B9bKZXAmjmrSp3OoLy5--oAiEOAVCRYvnOtRSKKlXo8)
> 
> I used Arduino IDE to compile the code... I'm very busy now, but tomorrow I will test it in my own ESP32-CAM module

like i said i try with 2 different code, the official one and [esp32-cam-webserver](https://github.com/easytarget/esp32-cam-webserver) they both detect the modul as 5640 and output a video so my guest its something wrong with this code. oh btw the screenshot you show its not even for this code, the one on the screenshot it the official one from the esp32 lib

GoostYTQC, 2024-05-21:
YES!!!!! I FIX IT

Sensor PID: 0x5640
Camera Model: 
[00:00:01.915 prepCam] Camera init OK for model OV5640 on board CAMERA_MODEL_AI_THINKER

i had to change :  `#define FB_CNT 4 // number of frame buffers` in the file mjpeg2sd to `#define FB_CNT 2 // number of frame buffers`

closed 2024-05-21 by GoostYTQC

davemaster, 2024-05-21:
Camera issue, don't forget a picture (better a video) of your success