s60sc/ESP32-CAM_MJPEG2SD #441: Heartbeat request to external URL

alojzjakob, 2024-05-27:
This would be awesome for number of reasons, specifically in my use case I cant have DDNS set up, so I would like to have my cameras periodically do a POST or just a GET request on defined URL, so I can then get the remote IP and keep my list updated and so I can access cameras whatever the IP they might get dynamically.

It would be awesome if it could be a POST with a JSON body, with camera info inside, most importantly the HTTP port used to access the web interface, because I will have multiple cameras on the same external IP, with different port forwarded ports through the 4g modem (mifi).

Would be cool to see HTTP port setting on the web interface too, so I dont need to flash each ESP32-cam with modifications to the global variables in the arduino code...

Thank you very much adn keep up the good work!

s60sc, 2024-05-27:
I'll take a look

s60sc, 2024-05-30:
Create a function like getExtIP() which connects to remote IP, uses HTTP client to send GET or POST then closes connnection. Create json string to send beforehand - see eg buildJsonString()

Call new function from doAppPing() as this is invoked every 30 secs (as set by wifiTimeoutSecs)

Once you have something working to your satisfaction I can incorporate it

alojzjakob, 2024-05-30:
Thank you for the directions!
I will try and do it soon, will let you know if I hit any problems :)

closed 2024-06-01 by s60sc