s60sc/ESP32-CAM_MJPEG2SD #444: Failed to compile with "Compilation error: invalid conversion" or "has no non-static data member"

kaihchang, 2024-05-30:
Following is part of error message. I'm not really sure where the main problem is, but I did successfully compile the exact same code few weeks ago, and I can't really remember what I've changed between.
Problem's probably in "jpg_reader_cb"?

Sometimes I also receive this error when trying to compile:
Compilation error: 'esp_mqtt_client_config_t' has no non-static data member named 'event_handle'

I guess there might be something I did wrong in the process, but I've checked everything I suspected.

---------------------------------------start of error message--------------------------------------------------------

FQBN: esp32:esp32:esp32:PartitionScheme=huge_app,PSRAM=enabled,EraseFlash=all Using board 'esp32'

...motionDetect.cpp:23:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
...ESP32-CAM_MJPEG2SD\motionDetect.cpp: In function 'bool jpg2rgb(const uint8_t*, size_t, uint8_t**, jpg_scale_t)':
...ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\motionDetect.cpp:378:50: error: invalid conversion from 'uint32_t (*)(void*, size_t, uint8_t*, size_t)' {aka 'long unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)'} to 'jpg_reader_cb' {aka 'unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)'} [-fpermissive]
  378 |   esp_err_t res = esp_jpg_decode(src_len, scale, _jpg_read, _rgb_write, (void*)&jpeg);
      |                                                  ^~~~~~~~~
      |                                                  |
      |                                                  uint32_t (*)(void*, size_t, uint8_t*, size_t) {aka long unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)}
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/espressif__esp32-camera/conversions/include/img_converters.h:25,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/espressif__esp32-camera/driver/include/esp_camera.h:250,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\appGlobals.h:181:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/espressif__esp32-camera/conversions/include/esp_jpg_decode.h:37:71: note:   initializing argument 3 of 'esp_err_t esp_jpg_decode(size_t, jpg_scale_t, jpg_reader_cb, jpg_writer_cb, void*)'
   37 | esp_err_t esp_jpg_decode(size_t len, jpg_scale_t scale, jpg_reader_cb reader, jpg_writer_cb writer, void * arg);
      |                                                         ~~~~~~~~~~~~~~^~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/i2s.h:23,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\globals.h:17,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\appGlobals.h:6,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\ESP32-CAM_MJPEG2SD.ino:8:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\ESP32-CAM_MJPEG2SD.ino: In function 'void setup()':
C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\ESP32-CAM_MJPEG2SD.ino:18:43: error: 'esp_spiram_get_size' was not declared in this scope
   18 |         LOG_INF("PSRAM size: %s", fmtSize(esp_spiram_get_size()));
      |                                           ^~~~~~~~~~~~~~~~~~~
C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\globals.h:349:61: note: in definition of macro 'LOG_INF'
  349 | #define LOG_INF(format, ...) logPrint(INF_FORMAT(format), ##__VA_ARGS__)
      |                                                             ^~~~~~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/i2s.h:23,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\globals.h:17,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\appGlobals.h:6,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\certificates.cpp:40:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/driver/deprecated/driver/i2s.h:23,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\globals.h:17,
                 from C:\Users\User\Desktop\hobbyist_projects\Arduino\ESP32-cam_video_stream_server\ESP32-CAM_MJPEG2SD\appGlobals.h:6,
[... cut ...]
Using library NetworkClientSecure at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\NetworkClientSecure 
Using library Wire at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\Wire 
exit status 1

Compilation error: invalid conversion from 'uint32_t (*)(void*, size_t, uint8_t*, size_t)' {aka 'long unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)'} to 'jpg_reader_cb' {aka 'unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)'} [-fpermissive]

s60sc, 2024-05-30:
as per readme, not yet compatible with arduino-esp32 v3.x

kaihchang, 2024-05-31:
That's likely the exact reason! Fell back to version 2.x and compiled successfully. Only problem I now have is sketch memory exceeds 4mb, but it should be managable.
Thx

closed 2024-05-31 by kaihchang