s60sc/ESP32-CAM_MJPEG2SD #446: [FEATURE UPDATE - TESTING] External heartbeat

alojzjakob, 2024-06-03:
I invite everyone to test external heartbeat feature by using www.espsee.com.
It is a website built with wordpress and a custom plugin which:
- handles heartbeat payloads
- automatically adds your cameras if you used token provided under your espsee user account
- displays all your cameras with up to date (dynamic) IP addresses
- shows camera active/incative status based on last heartbeat receive time

After a successful test i plan to opensource the plugin and we can also make epseee.com a default value in the camera settings screen

[image]

[image]

s60sc, 2024-06-04:
Youre welcome to add a reference. 
You should add a privacy policy.

alojzjakob, 2024-06-05:
Added EspSee as a hinted setting in external heartbeat feature's option.
Added explanation to the main readme.

Also added initial privacy policy: https://www.espsee.com/privacy-and-terms/, if you have any suggestions, please tell me :)
Added: https://www.espsee.com/about/

alojzjakob, 2024-06-07:
@s60sc as promised: https://github.com/alojzjakob/EspSee

cheers!

s60sc, 2024-06-07:
added link to readme

closed 2024-07-12 by s60sc