s60sc/ESP32-CAM_MJPEG2SD #448: I2S microphone conflicts with GPIO0

AYANAMl, 2024-06-06:
Hi, I want to use GPIO0 as a key input while testing your program. But when I turned on the microphone, the I2S bus was initialized, and a clock signal with a frequency of about 8MHz appeared on GPIO0, causing the level to remain at 0 when I read GPIO0.
At first, I thought it was crosstalk on the CLK signal line of the microphone, but the interference signal was too strong. Therefore, I analyzed your code and found that some things were missing when I2S allocated pins:
In audio.cpp file, your code is:

```
static esp_err_t setupMic() {
  // setup microphone based on its type, default I2S
  int i2sMode = I2S_MODE_MASTER | I2S_MODE_RX;
  i2s_mic_pins.bck_io_num = micSckPin; 
  i2s_mic_pins.ws_io_num = micSWsPin;
  i2s_mic_pins.data_in_num = micSdPin;
  i2s_mic_pins.data_out_num = I2S_PIN_NO_CHANGE;
```

When allocating I2S pins, the MCLK pin was missed, so it was assigned to GPIO0, which resulted in GPIO0 being unable to be used as a key. So I suggest modifying the code as follows:
```
static esp_err_t setupMic() {
  // setup microphone based on its type, default I2S
  int i2sMode = I2S_MODE_MASTER | I2S_MODE_RX;
  i2s_mic_pins.mck_io_num = -1;
  i2s_mic_pins.bck_io_num = micSckPin; 
  i2s_mic_pins.ws_io_num = micSWsPin;
  i2s_mic_pins.data_in_num = micSdPin;
  i2s_mic_pins.data_out_num = I2S_PIN_NO_CHANGE;
```

At this point, the MCLK pins will be cancelled and will not be placed on any pins. GPIO0 can be used normally.
Thank you!

AYANAMl, 2024-06-07:
Thank you for your reply. 

When configuring pins in the webpage, the SCK, SW, and SD pins of the I2S microphone can be configured. The MCK pins of I2S were not mentioned. The AI-THINKER board did indeed use GPIO0 as the camera's XCLK pin and did not use a microphone, while other development boards almost did not use GPIO0 pins. I am using XIAO ESP32S3, which includes a microphone. If the I2S microphone is initialized according to the current code, the GPIO0 pin will be automatically configured as the MCLK pin of I2S, and a clock signal will appear on that pin. Therefore, if we expect the GPIO0 pin to be used together with the microphone, such as setting GPIO0 as the lamp pin (ws2812) in the webpage, then I think we should make adjustments to the audio.cpp code. Because the current code will first initialize GPIO0 to a certain function (in the prepPeripheres() function), and then immediately initialize GPIO0 to the MCLK function of I2S (in the prepAudio() function), and the pre-set GPIO0 function will disappear.

Of course, if the function definition is strictly based on the existing peripherals of each development board, there will be no problems. If someone expects to add additional peripherals to their board or use this project in their own designed PCB, they may encounter this issue. This is just a suggestion, hoping to be helpful to those who are developing more features based on this project.

s60sc, 2024-06-07:
My initial response was based on the assumption that a lot of people raise issues without a proper investigation. You have done a proper investigation which I have now checked so I will add the change. Thanks.

closed 2024-07-12 by s60sc