s60sc/ESP32-CAM_MJPEG2SD #449: URL for a still-image?

vermoi63, 2024-06-16:
Dear developer, this is really great, great work, Tks for sharing this!!

Is there a way to have a URL that is capturing only a still-image in an external browser?
(background: want to put it in Home Assistant as a generic camera, or grap it with a "wget" request...)
I found the one for streaming, but not the one for still images without using the internal WEB Menu.

Kind regards!

s60sc, 2024-06-16:
Thanks

```
// Direct access URLs for NVR:
// - Video streaming: app_ip/sustain?video=1 
// - Audio streaming: app_ip/sustain?audio=1
// - Subtitle streaming: app_ip/sustain?srt=1
// - Stills: app_ip/control?still=1
```

vermoi63, 2024-06-16:
Tks, really really great....the still one simply not listed in documentation....just integrated in Home Assistant for 10 sec stills in the picture-glance and works great in parallel to the native firmware function!!!!

vermoi63, 2024-06-16:
Many Thanks for the hint to the still-image URL!

closed 2024-06-16 by vermoi63