s60sc/ESP32-CAM_MJPEG2SD #455: HTTPS file sending example

bjesuslopezg, 2024-07-02:
Hello, I've been using this app for a while to record videos using an ESP32S3 MCU, video recording works fine, however I have this particular requirement in which I need to send the recorded videos to an HTTPS server. It is of my understanding that this app supports HTTPS video sending but it seems to lack some examples on how to achieve it.

I located the file ftp.cpp, and I've already changed the mode from FTP to HFS using the web interface

[image]

However, I'm still really clueless on how to connect to the server and send the files. I believe it has something to do with these macros on the ftp.cpp file

[image]

s60sc, 2024-07-02:
Fill in the FS details on the web page and press File Upload button. The HTTPS server needs to process the http post message sent.

GerardoPrototype, 2024-07-03:
> Fill in the FS details on the web page and press File Upload button. The HTTPS server needs to process the http post message sent.

Hello, before asking, I wanted to congratulate you on the enormous work you contribute, thank you very much.

I had the same question about whether it is possible to send HTTP POST requests without having to specify the certificate. I created a server with an endpoint that receives a file through form-data, and the file is called "video."

I wanted to ask where exactly you have to specify the file name and where to put the URL.
Apologies if this part is not entirely clear to me.

[image]

s60sc, 2024-07-03:
Thanks
The url is built automatically from the info entered on the web page and the file or folder name selected from the drop down list. You need to provide certs for the esp in the certificates.cpp file to use https. The https server public cert is optional if youre not worried about server spoofing.

closed 2024-08-23 by s60sc