s60sc/ESP32-CAM_MJPEG2SD #456: Only I2S devices supported on I2S_NUM_1 - XIAO sense

freakontrol, 2024-07-05:
Hi, I'm having this issue where the audio is not working with the XIAO esp32s3 sense board. 
I tried to follow carefully the README.

In the web config interface I see.  
  
[image]   
  
I tried to change `PDM CLK pin -> 42` and `PDM DAT pin -> 41` it and went back to the original config after reboot.  
  
[image]  

I can see in the log `[WARN prepAudio] Only I2S devices supported on I2S_NUM_1` and `[WARN startAudio] Reboot ESP to use audio` after I trigger a recording.  
All recording are without audio.  
I also set the mic gain to 5.

```
00:00:00.009 logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.010 logSetup] =============== ESP-CAM_MJPEG 9.8.1 ===============
[00:00:00.020 logSetup] Compiled with arduino-esp32 v3.0.2
[00:00:00.020 printResetReason] Power on reset
[00:00:00.030 printWakeupReason] Wakeup by reset
[00:00:00.251 infoSD] SD card type SDHC, Size: 59.5GB
[00:00:00.252 listFolder] Sketch size 1.6MB
[00:00:00.413 listFolder] File: /data/configs.txt, size: 6188 bytes
[00:00:00.415 listFolder] File: /data/common.js, size: 38524 bytes
[00:00:00.418 listFolder] File: /data/MJPEG2SD.htm, size: 91KB
[00:00:00.429 listFolder] SD_MMC: 41.5MB used of 59.0GB
[00:00:00.429 loadConfig] Load config
[00:00:00.612 updateAppStatus] Enabling motion detection
[00:00:00.616 setup] PSRAM size: 8.0MB
[00:00:00.785 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_XIAO_ESP32S3
[00:00:00.973 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.004 WARN startWifi] SSID not connected not available, use AP
[00:00:01.005 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_D838180CDADC started, use 'http://192.168.4.1' to connect
[00:00:01.027 setupMdnsHost] mDNS service: http://seeed.local
[00:00:03.841 startPing] Started ping monitoring - On
[00:00:03.844 startWebServer] Starting web server on port: 80
[00:00:03.844 startWebServer] Remote server certificates not checked
[00:00:03.854 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:03.855 prepUpload] File uploads will use FTP server
[00:00:03.867 WARN prepAudio] Only I2S devices supported on I2S_NUM_1
[00:00:03.868 prepTelegram] Telegram not being used
[00:00:03.890 updateAppStatus] Enabling motion detection
[00:00:04.001 prepRecording] To record new AVI, do one of:
[00:00:04.001 prepRecording] - press Start Recording on web page
[00:00:04.001 prepRecording] - move in front of camera

[00:00:04.012 prepRecording] Camera model OV2640 on board CAMERA_MODEL_XIAO_ESP32S3 ready @ 20MHz
[00:00:04.022 checkMemory] Free: heap 163144, block: 147444, min: 163124, pSRAM 5662212
[00:00:04.022 loop] =============== Total tasks: 18 ===============

[00:00:06.252 isNight] Night time
[00:00:08.658 isNight] Day time
[00:00:08.842 resetWatchDog] WatchDog started using task: ping
[00:00:10.429 onWiFiEvent] WiFi AP client connection
[00:00:18.823 wsHandler] Websocket connection: 52
[09:08:38.022 showLocalTime] Got current time from browser: 05/07/2024 09:08:38 with tz: GMT0
[09:08:39.835 wsHandler] Websocket connection: 52
[09:08:41.485 appSpecificWebHandler] JPEG: 16661B in 157ms
[09:09:41.916 processFrame] Capture started by Button
[09:09:42.197 WARN startAudio] Reboot ESP to use audio
..................................................
..................................................
.....
[09:09:47.601 closeAvi] ******** AVI recording stats ********
[09:09:47.601 closeAvi] Recorded /20240705/20240705_090942_SVGA_20_5_S.avi
[09:09:47.611 closeAvi] AVI duration: 5 secs
[09:09:47.612 closeAvi] Number of frames: 105
[09:09:47.612 closeAvi] Required FPS: 20
[09:09:47.622 closeAvi] Actual FPS: 19.6
[09:09:47.622 closeAvi] File size: 1.6MB
[09:09:47.623 closeAvi] Average frame length: 15657 bytes
[09:09:47.633 closeAvi] Average frame monitoring time: 3 ms
[09:09:47.633 closeAvi] Average frame buffering time: 0 ms
[09:09:47.644 closeAvi] Average frame storage time: 43 ms
[09:09:47.644 closeAvi] Average SD write speed: 348 kB/s
[09:09:47.654 closeAvi] File open / completion times: 280 ms / 34 ms
[09:09:47.654 closeAvi] Busy: 98%
[09:09:47.665 checkMemory] Free: heap 166708, block: 147444, min: 155372, pSRAM 5201848
[09:09:47.665 closeAvi] *************************************
[09:09:47.676 checkFreeStorage] Storage free space: 59.0GB
[09:09:58.977 downloadFile] Download file: 20240705_090942_SVGA_20_5_S.avi, size: 1.6MB
```

s60sc, 2024-07-05:
as per readme, pdm mic not available when compiled with arduino v3.x, update should be ready in a few days

freakontrol, 2024-07-05:
Sorry I missed that info from the readme. Thank you for this project, I think is the most complete openly available!

freakontrol, 2024-08-08:
I cannot get the mic to work, the log file does not show any warning, but all the recording are mute. Also there's the same behaviour as in the first post, I try to change the default mic pinout and it went back to the original after reboot (I save it).
Maybe my board is faulty...

rjsachse, 2024-08-12:
Look in camera pins header, and change the predefined pi s there for you board.

closed 2024-08-23 by s60sc