s60sc/ESP32-CAM_MJPEG2SD #458: [Enhancement] Differentiate between PIR alert and Motion Detect alert in Telegram message

EERomeo, 2024-07-06:
Hey, thank you for the time you have put into this!

I have it working pretty well with Telegram alerts. I wanted to differentiate between a Motion Detect alert and a PIR alert on my Telegram messages. The reason for this would be if there is no light available, only the PIR would trigger. Otherwise the Motion Detection works very well. Also, you could just use a switch instead of the PIR sensor and know that physically the door is opened.

I would like to append at the end of the Telegram alert message something like "by PIR" or "by MD" depending of what the initiator was.

Can you give me some hints for where I would insert this?

Thanks!

s60sc, 2024-07-06:
Thanks
the call of tgramAlert() in closeAvi() is where you could customise

EERomeo, 2024-07-07:
Thank you, I'll look into it.

closed 2024-08-23 by s60sc