s60sc/ESP32-CAM_MJPEG2SD #465: nothing works

Atharvamaj, 2024-07-26:
so nothing work so first i put copied the data folder to the SD card and put it in the esp32 cam then i opened the program files in the ide esp32 dev module then enabled psram, then chose huge app also tried Minimal SPIFFS did not work then i flashed the board by pressing and holding the Io0 butten and pressing upload this i got this (15:53:29.542 -> ҊZ�%�禈!1�հ�~m1ֈ�	1���)�h�)N���(�������)i�!�ʎ�Y��1
�)Y!��= � !S����)�z��I(|�a!���!]��)i��)!Z�1E�������)) in the serial monitor. also i followed a video i got the first temp ip to work but now that does not work any more. i got this other error in the mqtt.cpp file
(C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp: In function 'void startMqttClient()':
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:182:119: error: 'esp_mqtt_client_config_t' has no non-static data member named 'event_handle'
  182 |   esp_mqtt_client_config_t mqtt_cfg{.event_handle = NULL, .host = "", .uri = mqtt_uri, .disable_auto_reconnect = false};
      |                                                                                                                       ^
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:183:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'username'
  183 |   mqtt_cfg.username = mqtt_user;
      |            ^~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:184:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'password'
  184 |   mqtt_cfg.password = mqtt_user_Pass;
      |            ^~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:185:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'client_id'
  185 |   mqtt_cfg.client_id = hostName;
      |            ^~~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:186:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'lwt_qos'
  186 |   mqtt_cfg.lwt_qos = MQTT_LWT_QOS;
      |            ^~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:187:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'lwt_msg'
  187 |   mqtt_cfg.lwt_msg = "offline";
      |            ^~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:188:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'lwt_retain'
  188 |   mqtt_cfg.lwt_retain = MQTT_LWT_RETAIN;
      |            ^~~~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\mqtt.cpp:192:12: error: 'esp_mqtt_client_config_t' {aka 'struct esp_mqtt_client_config_t'} has no member named 'lwt_topic'
  192 |   mqtt_cfg.lwt_topic = lwt_topic;
      |            ^~~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp: In function 'void prepServos()':
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:224:7: error: 'ledcSetup' was not declared in this scope; did you mean 'logSetup'?
  224 |       ledcSetup(SERVO_PAN_CHANNEL, PWM_FREQ, DUTY_BIT_DEPTH);
      |       ^~~~~~~~~
      |       logSetup
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:225:7: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  225 |       ledcAttachPin(servoPanPin, SERVO_PAN_CHANNEL);
      |       ^~~~~~~~~~~~~
      |       ledcAttach
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:228:7: error: 'ledcSetup' was not declared in this scope; did you mean 'logSetup'?
  228 |       ledcSetup(SERVO_TILT_CHANNEL, PWM_FREQ, DUTY_BIT_DEPTH);
      |       ^~~~~~~~~
      |       logSetup
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:229:7: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  229 |       ledcAttachPin(servoTiltPin, SERVO_TILT_CHANNEL);
      |       ^~~~~~~~~~~~~
      |       ledcAttach
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:234:5: error: 'ledcSetup' was not declared in this scope; did you mean 'logSetup'?
  234 |     ledcSetup(SERVO_STEER_CHANNEL, PWM_FREQ, DUTY_BIT_DEPTH);
      |     ^~~~~~~~~
      |     logSetup
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:235:5: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  235 |     ledcAttachPin(servoSteerPin, SERVO_STEER_CHANNEL);
      |     ^~~~~~~~~~~~~
      |     ledcAttach
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp: In function 'void setupLamp()':
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:414:7: error: 'ledcSetup' was not declared in this scope; did you mean 'logSetup'?
  414 |       ledcSetup(LAMP_LEDC_CHANNEL, 4000, 8);  // 12 kHz PWM, 8-bit resolution
      |       ^~~~~~~~~
      |       logSetup
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:415:7: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  415 |       ledcAttachPin(lampPin, LAMP_LEDC_CHANNEL);
      |       ^~~~~~~~~~~~~
      |       ledcAttach
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp: In function 'void stickTimer(bool)':
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:647:5: error: 'timerAlarmDisable' was not declared in this scope
  647 |     timerAlarmDisable(stickTimer);
      |     ^~~~~~~~~~~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\peripherals.cpp:653:28: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  653 |     stickTimer = timerBegin(2, 8000, true); // 0.1ms tick
      |                  ~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from C:\Users\athar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\cores\esp32/esp32-hal.h:84,
                 from C:\Users\athar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\cores\esp32/Arduino.h:36,
[... 51 lines cut ...]
  216 |     esp_task_wdt_init(wifiTimeoutSecs * 2, true); // panic abort on watchdog alert (contains wdt_isr)
      |                       ~~~~~~~~~~~~~~~~^~~
      |                                       |
      |                                       int
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\utils.cpp:216:22: error: too many arguments to function 'esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t*)'
  216 |     esp_task_wdt_init(wifiTimeoutSecs * 2, true); // panic abort on watchdog alert (contains wdt_isr)
      |     ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\utils.cpp:26:
C:\Users\athar\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/esp_system/include/esp_task_wdt.h:47:11: note: declared here
   47 | esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t *config);
      |           ^~~~~~~~~~~~~~~~~
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\utils.cpp: In function 'void initBrownout()':
C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\utils.cpp:1033:21: error: too few arguments to function 'esp_err_t rtc_isr_register(intr_handler_t, void*, uint32_t, uint32_t)'
 1033 |     rtc_isr_register(notifyBrownout, NULL, RTC_CNTL_BROWN_OUT_INT_ENA_M);
      |     ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\athar\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/driver/deprecated/driver/rtc_cntl.h:8,
                 from C:\Users\athar\OneDrive\Desktop\ESP32-CAM_MJPEG2SD\utils.cpp:1005:
C:\Users\athar\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/esp_hw_support/include/esp_private/rtc_ctrl.h:39:11: note: declared here
   39 | esp_err_t rtc_isr_register(intr_handler_t handler, void* handler_arg,
      |           ^~~~~~~~~~~~~~~~

exit status 1

Compilation error: 'esp_mqtt_client_config_t' has no non-static data member named 'event_handle') i need help i have this bord 
[image]

closed 2024-07-26 by s60sc

guece1, 2024-07-26:
An alternative installation process by [@ldijkman](https://github.com/ldijkman) is described [here](https://youtu.be/YLLGBM3i2aQ).

Browser functions only fully tested on Chrome.

Atharvamaj, 2024-07-28:
so that worked no error but i sill got this I don't I tried my network ssid and password did not wok.

Atharvamaj, 2024-07-28:
the photos 
[image]
[image]

Atharvamaj, 2024-07-28:
ones rebooted and when i connect to it, it does the same

bjesuslopezg, 2024-08-12:
Got what I believe were the same log errors in the console. I was using Arduino IDE 2.3.2, with arduino core 3.0.3. Went back to Arduino code 2.0.17 and everything was fixed. 

Take into account I'm working with a version of the project that may be like three months old.

Atharvamaj, 2024-08-13:
THANK YOU!! This helps me to finish my project.