s60sc/ESP32-CAM_MJPEG2SD #468: Startup Failure: Crash Loop Detected

uriukti, 2024-08-04:
Good afternoon. Why am I getting Startup Failure: Crash Loop Detected? on the web page. Your firmware is the latest version.

visionexbx, 2024-08-04:
I am having the same exact issue

s60sc, 2024-08-04:
what does the log say

visionexbx, 2024-08-04:
[00:00:03.021 logSetup] Setup RAM based log, size 7168, starting from 1300

[00:00:03.022 logSetup] =============== ESP-CAM_MJPEG 9.9.1 ===============
[00:00:03.032 WARN initBrownout] Brownout occurred due to inadequate power supply 
[00:00:03.032 logSetup] Compiled with arduino-esp32 v3.0.4
[00:00:03.043 printResetReason] Software reset due to brownout
[00:00:03.043 printWakeupReason] Wakeup by reset
[00:00:03.101 infoSD] SD card type SDHC, Size: 29.5GB
[00:00:03.101 listFolder] Sketch size 1.7MB
[00:00:03.502 listFolder] File: /data/MJPEG2SD.htm, size: 99KB
[00:00:03.505 listFolder] File: /data/common.js, size: 28713 bytes
[00:00:03.507 listFolder] File: /data/configs.txt, size: 6736 bytes
[00:00:03.517 listFolder] SD_MMC: 352KB used of 29.5GB
[00:00:03.518 loadConfig] Load config
[00:00:03.773 updateAppStatus] Enabling motion detection
[00:00:03.785 setup] PSRAM size: 4.0MB
[00:00:03.980 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:04.290 setWifiSTA] Wifi Station IP from DHCP
[00:00:04.325 WARN startWifi] SSID  not connected not available, use AP 
[00:00:04.329 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_408A48F7A608 started, use 'http://192.168.4.1' to connect
[00:00:04.340 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_408A48F7A608
[00:00:04.340 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_408A48F7A608 started, use 'http://192.168.4.1' to connect
[00:00:07.365 startPing] Started ping monitoring - On
[00:00:07.371 startWebServer] Starting web server on port: 80

visionexbx, 2024-08-04:
[00:00:07.371 startWebServer] Remote server certificates not checked
[00:00:07.371 WARN setup] Startup Failure: Crash loop detected 
[00:00:07.382 loop] =============== Total tasks: 14 ===============

[00:00:12.368 resetWatchDog] WatchDog started using task: ping

s60sc, 2024-08-04:
[00:00:03.032 WARN initBrownout] Brownout occurred due to inadequate power supply

visionexbx, 2024-08-05:
> [00:00:03.032 WARN initBrownout] Brownout occurred due to inadequate power supply

Thank you, tried several USB cables, finally found one that worked.

uriukti, 2024-08-05:
Yes.  I checked.  Power supply solved the problem.

uriukti, 2024-08-05:
It's all.

closed 2024-08-23 by s60sc