s60sc/ESP32-CAM_MJPEG2SD #470: New Board - ESP32-S3 CAM Development Board WiFi Bluetooth Module N16R8 +OV2640

mrmonteith, 2024-08-07:
I've had an issue getting a board to work.  So far not on your list, but seems to be pretty prevalent. 
[ESP32-S3 CAM Development Board WiFi Bluetooth Module N16R8 +OV2640](https://www.ebay.com/itm/145568550909?var=444710675402)

I did some tests and here is what I ran into with each:
CAMERA_MODEL_ESP32S3_EYE
SD Card and Camera work but gets hot

CAMERA_MODEL_ESP32S3_CAM_LCD
SD card doesn't work or camera, but cool
wifi works & web

CAMERA_MODEL_XIAO_ESP32S3
SD card doesn't work or camera, but cool
wifi works & web

CAMERA_MODEL_FREENOVE_ESP32S3_CAM
SD & Camera Works, but gets hot
wifi doesn't work

It took me a while to figure out their pinout.  A tiny graphic on the back with ~ _ and * for the pin types.  Here is the pins:
SD Card Pins
38,39,40

Camera Pins
4-13 

PSRAM Pins
35,36,37

If I find out more I'll update here.  I'm sure it's a matter of having the right pins mapped to the right functions and setting up a definition.

rjsachse, 2024-08-07:
looking at the pin out it is CAMERA_MODEL_FREENOVE_ESP32S3_CAM 
not sure why it would get hot. any logs

mrmonteith, 2024-08-08:
Not sure where logs are at.  I think there is a conflict between the pins in the CAMERA_MODEL_FREENOVE_ESP32S3_CAM pinouts and the SD card on this board.  This board the SD Card uses:
#define SD_MMC_CLK 39 
#define SD_MMC_CMD 38
#define SD_MMC_D0 40
 which are already used for other functions.

MartinBzm, 2024-08-09:
I have the same problem with a slightly different board, Ebay-link : https://www.ebay.com/itm/364956206606
It is difficult to get the firmware flashed as well. Following this topic with keen interest.

s60sc, 2024-08-09:
[MartinBzm](https://github.com/MartinBzm)  Try selecting CAMERA_MODEL_NEW_ESPS3_RE1_0

mrmonteith, 2024-08-11:
It's 1AM.  So hopefully this makes sense.  I tried the suggestion of using CAMERA_MODEL_NEW_ESPS3_RE1_0.  Here is the results:

As-Is, no changes

[00:00:00.029 logSetup] =============== ESP-CAM_MJPEG 9.9.1 ===============
[00:00:00.039 logSetup] Compiled with arduino-esp32 v3.0.4
[00:00:00.040 printResetReason] Power on reset
[00:00:00.050 printWakeupReason] Wakeup by reset
[00:00:00.079 WARN prepSD_MMC] SD card mount failed 
[00:00:00.244 setWifiSTA] Wifi Station IP from DHCP
.[00:00:00.411 onWiFiEvent] WiFi Station connection to mywifi, using hostname: 
...[00:00:01.930 onWiFiEvent] Wifi Station IP, use 'http://192.168.4.57' to connect
[00:00:02.079 onWiFiEvent] Station interface V6 IP addr is preferred
[00:00:02.645 onWiFiEvent] Wifi AP SSID: mywifi started, use 'http://192.168.4.1' to connect
[00:00:02.647 WARN setupMdnsHost] mDNS host:  Failed 
[00:00:04.079 onWiFiEvent] Station interface V6 IP addr is preferred
[00:00:05.660 startWifi] Wifi stats for mywifi - signal strength: -34 dBm; Encryption: WPA2_PSK; channel: 11
[00:00:05.661 startPing] Started ping monitoring - On
[00:00:05.673 startWebServer] Starting web server on port: 80
[00:00:05.674 startWebServer] Remote server certificates not checked
[00:00:05.684 WARN setup] Startup Failure: Check SD card inserted 
[00:00:05.684 loop] =============== Total tasks: 14 ===============

http://192.168.4.57
Wi-fi works, but get error: Startup Failure: Check SD card inserted

***********************************************************************
Changed SD card pins to:
#define SD_MMC_CLK 39 
#define SD_MMC_CMD 38
#define SD_MMC_D0 40
  Which is according to their pinouts.
  
[00:00:00.028 logSetup] =============== ESP-CAM_MJPEG 9.9.1 ===============
[00:00:00.038 logSetup] Compiled with arduino-esp32 v3.0.4
[00:00:00.038 printResetReason] Power on reset
[00:00:00.048 printWakeupReason] Wakeup by reset
[00:00:00.098 infoSD] SD card type SDHC, Size: 29.7GB
[00:00:00.099 listFolder] Sketch size 1.5MB
[00:00:00.243 listFolder] File: /data/common.js, size: 38896 bytes
[00:00:00.246 listFolder] File: /data/MJPEG2SD.htm, size: 92KB
[00:00:00.249 listFolder] File: /data/configs.txt, size: 6736 bytes
[00:00:00.260 listFolder] SD_MMC: 9.0MB used of 29.7GB
[00:00:00.260 loadConfig] Load config
[00:00:00.457 updateAppStatus] Enabling motion detection
[00:00:00.461 setup] PSRAM size: 8.0MB
[00:00:01.062 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.086 WARN startWifi] SSID  not connected not available, use AP 
[00:00:01.088 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_48A00CA28DCC started, use 'http://192.168.4.1' to connect
[00:00:01.109 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_4.local
[00:00:04.321 startPing] Started ping monitoring - On
[00:00:04.324 startWebServer] Starting web server on port: 80
[00:00:04.324 startWebServer] Remote server certificates not checked
[00:00:04.334 WARN setup] Startup Failure: Camera init error 0x105 on CAMERA_MODEL_NEW_ESPS3_RE1_0 
[00:00:04.345 loop] =============== Total tasks: 14 ===============

SD Card works, but Wi-fi doesn't.
******************************
  I pushed to try to find a schematic.  No luck at that.  I will keep digging.  Trying to make heads or tails out of it.   Seems to be a conflict.   

Now here is a note. I can use the sketch called ESP32 CameraWebserver works.  Then again it doesn't use the SD card.  It gets very hot.  But heard that it's a power issue.   If we figure it out great.  If not I already have a XIAO Sense ordered.  I'll play with this until it arrives.   If you come up with something I'll be glad to test it.

s60sc, 2024-08-11:
CAMERA_MODEL_NEW_ESPS3_RE1_0 was for [MartinBzm](https://github.com/MartinBzm), you need to use CAMERA_MODEL_FREENOVE_ESP32S3_CAM
If you still have problems it is a hardware issue

mrmonteith, 2024-08-12:
I tried the CAMERA_MODEL_FREENOVE_ESP32S3_CAM.  I even changed the SD card pins to match what they said they were for that board. I can get each of the components to work separately testing each.  But they just will NOT all work together.   Thanks for trying to help.  But I've wasted enough time on this.  Waiting for some other, more widely used type boards.  Thanks

closed 2024-08-23 by s60sc