s60sc/ESP32-CAM_MJPEG2SD #471: Camera not working with DFRobot FireBeetle 2 ESP32-S3 Board

maximagination1, 2024-08-18:
Hi,

So, I received x2 new DFRobot FireBeetle 2 ESP32-S3 boards that I found to compile your code by not running it due to the error "camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)". I can confirm this is not a hardware issue. I ran the example CameraWebServer code with my modifications having an accompanying library from DFRobot that made the camera work and open the basic web server.

To make this board's camera work, I uncovered that it needs the DFRobot_AXP313A library to power on the camera sensor via the board's AXP313A chip. This camera power-on method seems different than any other ESP32 I've used.

You can read more about the library here: https://github.com/cdjq/DFRobot_AXP313A?tab=readme-ov-file

And the board's documentation: https://wiki.dfrobot.com/SKU_DFR0975_FireBeetle_2_Board_ESP32_S3

I've attempted including the library and rolling it out in your code (in `logSetup()`) with no avail.

If you know how it needs to be implemented, I'd appreciate knowing what steps to take so the following is included and works:

`#include "DFRobot_AXP313A.h"

DFRobot_AXP313A axp;

while(axp.begin() != 0){
    Serial.println("init error");
    delay(1000);
  }
axp.enableCameraPower(axp.eOV2640);//设置摄像头供电` 

Thanks,

Max

s60sc, 2024-08-18:
try adding the code to start of` prepCam()` in `mjpeg2sd.cpp`

ensure log shows `CAMERA_MODEL_DFRobot_FireBeetle2_ESP32S3` as the selected camera

maximagination1, 2024-08-18:
I tested that, but still get an error. No matter whether the code is placed in `setup(),` `prepCam(),` or `prepRecording(),` I'm met with a Startup crash loop failure and "WARN prepRecording] failed to get camera frame" error.

Connecting those errors to the source, I also seem to get a Guru Meditation error with a decoded stack result pointing to line 708 in `prepRecording()`

Any idea on what's missing?

Thanks.

s60sc, 2024-08-18:
Provide complete startup log

maximagination1, 2024-08-18:
Nevermind, but thanks anyway. The code is working and running fine on the FireBeetle board.

I had uploaded with 16MB Partition Scheme which might have been what was causing the issue. Instead selected 8MB with Spiffs, then it worked.

The working added initialization of the image sensor goes as follows:
`#include "DFRobot_AXP313A.h"` and `DFRobot_AXP313A axp;` at the top of `mjpeg2sd.cpp`

 and

`while (axp.begin() != 0) {
    Serial.println("init error");
    delay(1000);
  }
  axp.enableCameraPower(axp.eOV2640);//设置摄像头供电`

sits close at the top of `prepCam,` but below `framesize_t` code as shown here: `
[image]
`

It may help others with FireBeetle 2 boards to update your code with these changes.

Thanks anyway!

s60sc, 2024-08-18:
Will do.  Does this model have no SD card slot?

maximagination1, 2024-08-18:
This model doesn't have one, no. Had to upload the app via SPIFFS.

s60sc, 2024-08-23:
support added to v9.9.4

closed 2024-08-23 by s60sc