s60sc/ESP32-CAM_MJPEG2SD #472: [Module Support Inquiry] SDMMC MicroSD Card Support for FireBeetle 2 ESP32-S3

maximagination1, 2024-08-25:
Hi,

I purchased an SDMMC MicroSD card module as you recommended. It uses pins CD, CS, DI, DO, and CLK. How can it be interfaced with an ESP32 such as the FireBeetle board using your code?

I don't see pin defines for the FireBeetle in your code that indicate an SD card can connect to it. Are there some sort of default pins the ESP32-S3 board has that are designed for an external SDMMC module? If so, which would those be? Thanks.

Best regards,

Max

s60sc, 2024-08-25:
Choose any 3 free pins for CLK, CD and D0 and add the code below to the firebeetle entry with the relevant pin numbers, 
in place of `#define NO_SD `

```
#define SD_MMC_CLK CLK pin
#define SD_MMC_CMD CD pin
#define SD_MMC_D0  D0 pin
```

All 3 pins need 10k pullups which should be on the SD module
[info](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/README.md)

maximagination1, 2024-08-26:
Thanks for your prompt response.

I did as you mentioned, but I'm getting the "Startup Failure: Crash Loop Detected" error. What could this be?

s60sc, 2024-08-26:
Hardware issue, either with pin assignment or lack of pullups

maximagination1, 2024-08-26:
I checked my hardware, connections are firm and short between the 3.3V-powered SD module and my ESP32S3. I've formatted the micro-SD card with FAT32 and 8192 bytes - not sure if these details determine whether the SD card will be read. I've tested the module with pull-up resistors on those 3 pins, still no luck...

Supposedly, this SD card module I am using doesn't seem to need resistors as it has a logic level shifter, according to this page on Adafruit here: https://learn.adafruit.com/adafruit-micro-sd-breakout-board-card-tutorial/look-out

This is the module I'm using: https://www.adafruit.com/product/254

I assume this is SDMMC compatible. If it needs to be connected up some other way or if one or more pins need to be attached, I'd appreciate it if you could let me know.

Thanks.

s60sc, 2024-08-26:
Over complicated board, you need to connect 3V3, 5V and GND to power the level shifter.

maximagination1, 2024-08-26:
Yes, it's getting too complicated using this module.

How about using a Micro-SD card adapter instead of the previous module, would it work directly with the ESP32S3? Like one of these:
[image]

s60sc, 2024-08-26:
Best guess

pin 3 CMD pullup
pin 4 3V3
pin 5 CLK
pin 6 GND
pin 7 D0 pullup

for each pullup, connect 10k resistor between pin and 3V3

pins 1, 2, 8 may also benefit from pullups to stop them floating

s60sc, 2024-08-26:
pins given for micro sd

maximagination1, 2024-08-26:
Ok, thanks! Will test this out.

maximagination1, 2024-08-26:
I got the SD card adapter working. Quite a fast and cheap solution.

I pulled-up all four data pins and the one command pin with a total of five 10K ohm resistors and adjoined both GNDs (VSS1 & VSS2). 

Works perfectly!

Thank you very much for your help as always!

About the Hexapedal Robot code, I still have to build the robot and ensure everything works as expected before merging it with yours.

Cheers.

s60sc, 2024-08-26:
Good to hear, interested to see the end result

rjsachse, 2024-10-04:
I use a breakout board like this 
[image]
Just need to make sure it is sd mmc not spi like most on the market

closed 2025-01-22 by s60sc