s60sc/ESP32-CAM_MJPEG2SD #480: [FEATURE REQUEST] ONVIF support?

TheWISPRer, 2024-09-15:
I was wondering if y'all had tried adding ONVIF support so that these cameras can be added to common NVR platforms. It would allow greater flexibility and integration into larger systems using a standard protocol. 

I am trying to integrate this myself, but think that this is beyond my skillset. Figured I'd go ahead and open a feature request issue JIC. Will update if I am able to succeed in this or not.

s60sc, 2024-09-16:
could try using go2rtc to convert mjpeg/wav to onvif  https://github.com/s60sc/ESP32-CAM_MJPEG2SD#stream-to-nvr

TheWISPRer, 2024-09-17:
Well, looks like that will not fit in the use case examples of this request. ONVIF is really just a discovery protocol that provides endpoint information to the main unit from the camera. It typically utilizes an RTSP stream, but ONVIF can support direct MJPEG on most clients, so would not necessarily need conversion. That go2rtc library adds a bunch of nice, but unnecessary bloat for this particular application requirement.

Attempting to add WSD and SOAP so that it is natively discoverable through ONVIF without any middleware. Will update on progress. Definitely not my typical wheelhouse, so don't hold your breath.

rjsachse, 2024-10-05:
Ok so I had a quick look today. Yes might be possible. I done a quick test with ODM onvif device manager. The big issue is time incorporating all of the services and actions.. I'm working on other stuff  mainly hardware.  I can share what I done for a test using esp32 cam webserver. Does not currently stream as I haven't added all the functions for services and actions, but might be a starting point to help someone else out if they are willing too.

rjsachse, 2024-12-26:
Ok good news. I have implemented rtsp for video audio and now just finished subtitles. Also I have a working example of onvif too. Just need to clean up code and add webserver settings. Hopefully within the next few weeks. Stay tuned

TheWISPRer, 2024-12-28:
Awesome. I'll keep an eye out for it! I had a use-case for onvif come up a week ago and picked back up looking into this/trying to implement, but have not been able to get it to discover by ODM yet.. Probably something stupid I'm missing.

How does the RTSP performance look to you? Everything I've tried so far (without getting too deep in the weeds of things) has been subpar at best. Typically, unstable video speeds/framerates + desync issues due to encoding overhead. Frame limiting has had some success, but still not good enough performance to utilize outside of the bench.

rjsachse, 2025-01-03:
sorry for the late reply, ok so I have nearly finished the RTSP library and getting perfect FPS

QQVGA | 50 Fps
QCIF | 50  Fps
HQVGA | 50 Fps
240X240 | 50 Fps
QVGA | 50 Fps
CIF | 50 Fps
HVGA | 50 Fps
VGA | 25 Fps
SVGA | 25 Fps
XGA | 12.5 Fps
HD | 12.5 Fps
SXGA | 12.5 Fps
UXGA | 5 Fps

Now when adding to this project will only be as fast as it can record when recording but just streaming i was getting 25Fps With SVGA

rjsachse, 2025-01-04:
Good News Library is out [RTSPServer](https://github.com/rjsachse/RTSPServer)

rjsachse, 2025-02-01:
I have created a Discussion for ONVIF support. can close this

closed 2025-05-09 by TheWISPRer