s60sc/ESP32-CAM_MJPEG2SD #481: Brilliant programme

Donbruno100, 2024-09-22:
Your new V10. is awesome and welldone guys brilliant work, however on the peripherals did you leave out the slider "on" button for Lamp purpously, it just says Lamp pin and then Manual/Pir and not slider to activate it, or am I loosing my marbles??
Keep up the great work....

s60sc, 2024-09-22:
Thanks
lamp use slider removed as redundant as lamp level slider can switch off lamp

Donbruno100, 2024-09-22:
Thanks, appreciate your responce ..... only problem is its not going on like the previous version, somehow its not holding pin number ( 4 ) even on save and reboot and the lamp slider on Camera control is active but nothing goes on.... the previous version was updated immediately and worked so great... anyway if its redundant then its redundant....Thanks again for your responce and help buds...

s60sc, 2024-09-23:
after entering pin number, click focus away from pin field so that browser registers the input

closed 2024-10-05 by s60sc