s60sc/ESP32-CAM_MJPEG2SD #482: Lamp only ON for ~15 sec after change in level

vermoi63, 2024-09-22:
in latest version it seems the lamp turns off automatically after ~15 secs of a change in the lamp level
- is this wanted?
- where do I find the code as I want to switch a relay on pin4 that turns an external IR LED on for night

Thanks in advance!

s60sc, 2024-09-23:
lamp behaviour is unknown
to operate relay manually, set lamp level to min or max, but not in between.
to auto switch would need extra code

s60sc, 2024-09-23:
when timelapse in on and nighttime then lamp will come on for each timelapse photo

vermoi63, 2024-09-23:
what I described is in "day-mode".....here the lamp is put off every timelaps interval it seems.....but I would have expected that is is completely in manuel control......can this be changed, so that it is up to the manual control if lamp is on or off, not being disturbed by the timelaps interval.....

OR...is there a simple other pin I can use for manual control of ON/OFF to switch a relay by fully manual control?

Kind regards!

s60sc, 2024-09-23:
I'll add code for a relay pin in a couple of days

vermoi63, 2024-09-24:
GREAT THANKS!

s60sc, 2024-09-24:
relay parameters under Peripherals tab, use eg pin 12

vermoi63, 2024-10-01:
Issue solved by developer: by adding a relay pin/switch that can be switched independently from flash light/lamp!
(from Version 10.2.2 on)

Great Thanks!

closed 2024-10-05 by s60sc