s60sc/ESP32-CAM_MJPEG2SD #483: WLAN hangs up too often whean video streaming with fast changes in content

vermoi63, 2024-09-22:
WLAN hangs up (only ping watchdog) in video streaming always when lots of changes in picture occur.
-code V 9.9.4
-arduino board manager for espressif 3.0.4
-board esp32cam AI thinker

Any idea to get this stable?
(also occurs in CIF resolution with only 1 fps)

vermoi63, 2024-09-22:
to be added: happnes only when motion detect is ON AND doing streaming
(same without streaming and only recording to SD works fine)

s60sc, 2024-09-23:
streaming bandwidth entirely dependent on wifi signal quality

vermoi63, 2024-09-23:
Thank you for the response,
I found out that turning motion-detection OFF while downloading or streaming works well (no WLAN restart occurs);
so I assume simply the processor is not fast enough - at least this is a workaround for week CPU' (esp32)
Just ordered a S3 to see if it manages motion-detection AND download or streaming better

Thanks again for all the engineering you provide!

vermoi63, 2024-09-23:
ddd

closed 2024-09-23 by vermoi63