s60sc/ESP32-CAM_MJPEG2SD #485: Startup Failure: Crash loop detected, check log

daothanhluan92, 2024-09-24:
[image]
I need help! Thanks everyone!

s60sc, 2024-09-24:
show start of log

rjsachse, 2024-09-24:
Is that after flashing? Or after manually resetting?

gemi254, 2024-09-25:
Your sd card may be corrupted...!!
Try .
http://[ipaddress]/control?deldata=1
and setup your device again..

rjsachse, 2024-09-25:
Looks like sd card not formatted right or have other issues  so in browser address bar type `http://[ipaddress]/control?deldata=1` where `[ipaddress]` put you ip address in eg. `192.168.0.113` without the `[ ]` for Example `http://192.168.0.113/control?deldata=1`

sash85, 2024-10-01:
Unfortunately, I have the same problem. When I execute the command "control?deldata=1", everything works. Restarting within the user interface is also no problem. However, when I unplug the ESP and then plug it back in again, the same error occurs. I have completely reinstalled the software with the latest version (downloaded today). I have completely formatted the SD card.
I read in another issue, that this guy has a problem with PIR. I had also activated PIR, but left it off when reinstalling. One sensor is also not connected. I also tried a different power supply. I get another SD Card tomorrow and and hope that will solve the problem.

But maybe someone has another an idea?

LOG
[00:00:01.682 logSetup] Setup RAM based log, size 7168, starting from 3394

[00:00:01.683 logSetup] =============== ESP-CAM_MJPEG 10.2.2 ===============
[00:00:01.693 WARN initBrownout] Brownout warning previously notified 
[00:00:01.693 logSetup] Compiled with arduino-esp32 v3.0.5
[00:00:01.703 printResetReason] Software reset due to exception/panic
[00:00:01.704 printWakeupReason] Wakeup by reset
[00:00:01.882 infoSD] SD card type SDHC, Size: 7.5GB
[00:00:01.882 listFolder] Sketch size 1.5MB
[00:00:02.208 listFolder] File: /data/configs.txt, size: 7282 bytes
[00:00:02.211 listFolder] File: /data/common.js, size: 46093 bytes

[00:00:02.711 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:02.965 setWifiSTA] Wifi Station IP from DHCP
.[00:00:03.036 onWiFiEvent] WiFi Station connection to XXXX, using hostname: ESP-CAM_MJPEG_58F8B67A9834
[00:00:03.052 onWiFiEvent] Wifi Station IP, use 'http://192.168.XXX.XXX' to connect
[00:00:03.497 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_58F8B67A9834 started, use 'http://192.168.4.1' to connect
[00:00:06.518 startWifi] Wifi stats for XXXX - signal strength: -68 dBm; Encryption: WPA2_PSK; channel: 1
[00:00:06.519 startWifi] Wifi stats for XXXX - signal strength: -80 dBm; Encryption: WPA2_PSK; channel: 11
[00:00:06.530 startPing] Started ping monitoring - On
[00:00:06.843 wsHandler] Websocket connection: 52
[19:15:07.896 getLocalNTP] Using NTP server: pool.ntp.org
[19:15:07.897 showLocalTime] Got current time from NTP: 01/10/2024 19:15:07 with tz: GMT0
[19:15:37.897 setAlarm] Alarm scheduled at 02/10/2024 01:00:00
[19:15:38.683 getExtIP] External IP: 212.99.192.103
[19:15:38.685 doAppPing] Daily rollover
[19:19:37.897 isNight] Night time

s60sc, 2024-10-01:
[00:00:01.693 WARN initBrownout] Brownout warning previously notified

You need  a better power supply

sash85, 2024-10-02:
@s60sc Thank you for your fast reply! :)

I tried four different Power Supplies, including macOS USB, Power Bank, USB Power-Supply and Universal Adapter (all 5V and 2,4A, 4.5A and 5A). Furthermore, I tried a thicker cable and the other GND Pin.

I will try a different ESP-Cam later this day. Maybe this one is broken.

[06:55:23.868 logSetup] =============== ESP-CAM_MJPEG 10.2.2 ===============
[06:55:23.879 WARN initBrownout] Brownout warning previously notified
[06:55:23.879 logSetup] Compiled with arduino-esp32 v3.0.5
[06:55:23.889 printResetReason] Software reset via esp_restart
[06:55:23.889 printWakeupReason] Wakeup by reset
[06:55:24.396 infoSD] SD card type SDHC, Size: 7.5GB
[06:55:24.397 listFolder] Sketch size 1.5MB
[06:55:24.750 listFolder] SD_MMC: 8.0MB used of 7.5GB
[06:55:24.751 loadConfig] Load config
[06:55:24.825 checkConfigFile] Created /data/configs.txt from local store
[06:55:24.995 updateAppStatus] Enabling motion detection
[06:55:25.004 setup] PSRAM size: 4.0MB
[06:55:25.197 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[06:55:25.466 setWifiSTA] Wifi Station IP from DHCP
.[06:55:25.531 onWiFiEvent] WiFi Station connection to Teddy, using hostname: ESP-CAM_MJPEG_58F8B67A9834
[06:55:25.544 onWiFiEvent] Wifi Station IP, use 'http://192.168.122.186' to connect
[06:55:26.044 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_58F8B67A9834 started, use 'http://192.168.4.1' to connect
[06:55:29.064 startWifi] Wifi stats for Teddy - signal strength: -58 dBm; Encryption: WPA2_PSK; channel: 1
[06:55:29.065 startWifi] Wifi stats for Teddy - signal strength: -83 dBm; Encryption: WPA2_PSK; channel: 11
[06:55:29.076 startPing] Started ping monitoring - On
[06:55:29.093 resetWatchDog] WatchDog started using task: ping
[06:55:29.093 getLocalNTP] Using NTP server: pool.ntp.org
[06:55:29.094 startWebServer] Starting web server on port: 80
[06:55:29.094 showLocalTime] Got current time from NTP: 02/10/2024 06:55:29 with tz: GMT0
[06:55:29.105 startWebServer] Remote server certificates not checked
[06:55:29.116 startSustainTasks] Started 1 HTTP sustain tasks
[06:55:29.162 updateAppStatus] Enabling motion detection
[06:55:29.393 prepRecording] To record new AVI, do one of:
[06:55:29.393 prepRecording] - press Start Recording on web page
[06:55:29.393 prepRecording] - move in front of camera

[06:55:29.405 prepRecording] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[06:55:29.406 checkMemory] Free: heap 63320, block: 32756, min: 61084, pSRAM 1483912
[06:55:29.418 loop] =============== Total tasks: 16 ===============

[06:55:29.532 wgetFile] Downloading /data/common.js from /s60sc/ESP32-CAM_MJPEG2SD/master/data/common.js
[06:55:30.186 wgetFile] Downloaded /data/common.js, size 46093 bytes
[06:55:30.639 wgetFile] Downloading /data/MJPEG2SD.htm from /s60sc/ESP32-CAM_MJPEG2SD/master/data/MJPEG2SD.htm
[06:55:31.819 wgetFile] Downloaded /data/MJPEG2SD.htm, size 97KB
[06:55:59.084 setAlarm] Alarm scheduled at 03/10/2024 01:00:00
[06:55:59.898 saveConfigVect] Config file saved
[06:55:59.929 doAppPing] Daily rollover
[06:56:07.071 deleteFolderOrFile] Deleting : /data
[06:56:07.071 deleteFolderOrFile] Folder /data contents
[06:56:07.107 deleteFolderOrFile] FILE : /data/configs.txt Size : 7282 bytes deleted
[06:56:07.130 deleteFolderOrFile] FILE : /data/common.js Size : 46093 bytes deleted
[06:56:07.155 deleteFolderOrFile] FILE : /data/MJPEG2SD.htm Size : 97KB deleted
[06:56:07.178 deleteFolderOrFile] Folder /data deleted
[06:56:07.192 doRestart] Controlled restart: user requested restart after data deletion
[06:56:09.208 onWiFiEvent] WiFi Station disconnected
[06:56:09.209 onWiFiEvent] Wifi Station stopped Teddy
[06:56:10.230 logSetup] Setup RAM based log, size 7168, starting from 5718

[06:56:10.231 logSetup] =============== ESP-CAM_MJPEG 10.2.2 ===============
[06:56:10.241 WARN initBrownout] Brownout warning previously notified
[06:56:10.241 logSetup] Compiled with arduino-esp32 v3.0.5
[06:56:10.251 printResetReason] Software reset via esp_restart
[06:56:10.252 printWakeupReason] Wakeup by reset
[06:56:10.636 infoSD] SD card type SDHC, Size: 7.5GB
[06:56:10.636 listFolder] Sketch size 1.5MB
[06:56:10.960 listFolder] SD_MMC: 8.0MB used of 7.5GB
[06:56:10.961 loadConfig] Load config
[06:56:11.039 checkConfigFile] Created /data/configs.txt from local store
[06:56:11.180 updateAppStatus] Enabling motion detection
[... 5 lines cut ...]
[06:56:12.186 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_58F8B67A9834 started, use 'http://192.168.4.1' to connect
[06:56:15.208 startWifi] Wifi stats for Teddy - signal strength: -55 dBm; Encryption: WPA2_PSK; channel: 1
[06:56:15.208 startWifi] Wifi stats for Teddy - signal strength: -81 dBm; Encryption: WPA2_PSK; channel: 11
[06:56:15.220 startPing] Started ping monitoring - On
[06:56:15.235 resetWatchDog] WatchDog started using task: ping
[06:56:15.235 getLocalNTP] Using NTP server: pool.ntp.org
[06:56:15.235 startWebServer] Starting web server on port: 80
[06:56:15.236 showLocalTime] Got current time from NTP: 02/10/2024 06:56:15 with tz: GMT0
[06:56:15.246 startWebServer] Remote server certificates not checked
[06:56:15.257 startSustainTasks] Started 1 HTTP sustain tasks
[06:56:15.302 updateAppStatus] Enabling motion detection
[06:56:15.549 prepRecording] To record new AVI, do one of:
[06:56:15.550 prepRecording] - press Start Recording on web page
[06:56:15.550 prepRecording] - move in front of camera

[... cut ...]
[06:56:15.572 loop] =============== Total tasks: 16 ===============

[06:56:15.667 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm
[06:56:15.689 wgetFile] Downloading /data/common.js from /s60sc/ESP32-CAM_MJPEG2SD/master/data/common.js
[06:56:16.126 wgetFile] Downloaded /data/common.js, size 46093 bytes
[06:56:16.544 wgetFile] Downloading /data/MJPEG2SD.htm from /s60sc/ESP32-CAM_MJPEG2SD/master/data/MJPEG2SD.htm
[06:56:17.639 wgetFile] Downloaded /data/MJPEG2SD.htm, size 97KB
[06:56:23.238 wsHandler] Websocket connection: 53

closed 2024-10-05 by s60sc

sash85, 2024-10-21:
Hi again, apologies for the delayed response. I was ill and could not check it earlier. I loaded the new Version last Friday and used another esp32-cam. Unfortunately, I still have problems. It worked with a 4 GB and an 8 GB card (no names). However, I ordered 10 SanDisk 32 GB Micro SD cards (SanDisk Ultra 10 A1) for a school project that I do on a voluntary basis and none of them work with it. The cards themselves show no errors. I tested:
FAT32 with different clusters (8, 16, 32), one partition with 16 GB and one with 8 GB (each on a 32 GB card). I tried SD Card Formatter on Mac and Partition Wizard on Windows.
Do you have any other tips, or is it the cards? If so, are there any recommendations for other cards?

If I should open a new ticket, please write me. Thank you!

sash85, 2024-10-21:
Log:
00:00:00.474 logSetup] Setup RAM based log, size 7168, starting from 0

[00:00:00.475 logSetup] =============== ESP-CAM_MJPEG 10.4.2 ===============
[00:00:00.485 logSetup] Compiled with arduino-esp32 v3.0.5
[00:00:00.485 printResetReason] Power on reset
[00:00:00.496 printWakeupReason] Wakeup by reset
[00:00:00.524 WARN prepSD_MMC] SD card mount failed
[00:00:00.711 setWifiSTA] Wifi Station IP from DHCP
[00:00:00.717 WARN startWifi] SSID not connected not available, use AP
[00:00:03.734 startPing] Started ping monitoring - On
[00:00:03.737 startWebServer] Starting web server on port: 80
[00:00:03.737 startWebServer] Remote server certificates not checked
[00:00:03.748 WARN setup] Startup Failure: Check SD card inserted
[00:00:03.748 loop] =============== Total tasks: 12 ===============

s60sc, 2024-10-22:
There should not be an issue with 32gb cards so I dont have an answer for you

sash85, 2024-10-24:
Thank you for your time and your response! :-)

I'm starting to get a bit desperate.
The following is the status, maybe someone has an idea:
I upload the code using the Arduino IDE and insert a formatted micro SD card (without data folder) into the ESP32-Cam. Then I connect to the AP, enter the Wi-Fi data and can then connect and everything works. I save small adjustments and restart. Everything works. I unplug the device for a few seconds, everything works. I unplug it for a little longer and the SD card is no longer recognized.

I have tested ESPs from various series, cheap micro SD cards and two Intenso 8 GB.
I can't believe no one else has these problems.

Arduino IDE:

- ESP32 Dev Module
- Board: esp32 3.0.7 (also testet with 3.0.5)
- #define CAMERA_MODEL_AI_THINKER  is acitve
- #define INCLUDE_PERIPH true
- Minimal SPIFFS
- PSRAM Enabled
- Flash Size 4MB
- Downloaded Code from github yesterday
- 8 GB Intenso FAT32 with 32 Kb Cluster

LOG:
[00:00:00.488 logSetup] =============== ESP-CAM_MJPEG 10.4.2 ===============
[00:00:00.498 logSetup] Compiled with arduino-esp32 v3.0.7
[00:00:00.499 printResetReason] Power on reset
[00:00:00.509 printWakeupReason] Wakeup by reset
[00:00:00.537 WARN prepSD_MMC] SD card mount failed
[00:00:00.750 setWifiSTA] Wifi Station IP from DHCP
[00:00:00.756 WARN startWifi] SSID not connected not available, use AP
[00:00:04.173 startPing] Started ping monitoring - On
[00:00:04.176 startWebServer] Starting web server on port: 80
[00:00:04.176 startWebServer] Remote server certificates not checked
[00:00:04.187 WARN setup] Startup Failure: Check SD card inserted
[00:00:04.187 loop] =============== Total tasks: 12 ===============

[00:00:09.173 resetWatchDog] WatchDog started using task: ping
[00:09:39.091 onWiFiEvent] WiFi AP client connection
[00:09:50.375 wsHandler] Websocket connection: 52
[07:44:38.000 showLocalTime] Got current time from browser: 24/10/2024 07:44:38 with tz: GMT0

s60sc, 2024-10-24:
INCLUDE_PERIPH true - have you defined or connected any pins

sash85, 2024-10-24:
I've connected a PIR Sensor to Pin 12 and setup everything (turn on light, record video). It works perfect (also after restart, i recorded and streamed video, the light turn on in the dark).

But i also setup devices without connecting or set the PIR to check if that is the bug. Should i try it with “INCLUDE_PERIPH false”? 

I also tried the Webinstaller without SD Card from this page. This device works great, but I need the SD card to save files.
https://ldijkman.github.io/async-esp-fs-webserver/ino/ESP32-CAM_Camera_Webserver/BIN/CAM-Flash.html

s60sc, 2024-10-24:
Readme states:
pin 12: Only use as output pin.
pin 12 is a strapping pin, which must be low during startup which is not guaranteed with a PIR input, unless you disable the strapping function by burning an efuse

sash85, 2024-10-24:
Ok, thank you very much, then I read that wrong. Just to be on the safe side, I tried it again with an esp32-cam that didn't have a PIR sensor connected. The problem still exists. :-/

Maybe i bought bad hardware with a bad connection so the sd card.

What I don't understand is why it works without any problems at first and then not the next time.

sash85, 2024-11-19:
Ok, i tested some micro SD Cards and found out that all cards with “SD HC 1 I” are working (also 32 GB Cards).
Cards with U1 (UHS Speed Class 1) and UHS-I (UHS Bus Speed) are fine. Cards without this Symbols are not.
More info here: https://bulkmemorycards.com/what-the-symbols-on-an-sd-card-mean/

I ordered these cards, and they are working fine for the first time: Netac Micro SD Karte 32gb Fat32 
I haven't been able to do a long-term test yet. If I find any problems, I'll write here again.

For the PIR is here a solution: https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/501#issuecomment-2395063750