s60sc/ESP32-CAM_MJPEG2SD #487: undefined reference to `relayPin' and `relayMode'

daniel-frenkel, 2024-09-25:
When compiling using ESP ARDUINO 3.0.4, I receive the following errors, resulting in failure:

```
undefined reference to `relayPin'
undefined reference to `relayMode'
```

Looks like it may be related to the 10.2.1 update

deanfourie1, 2024-09-26:
Same for me.

```
/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: 
\AppData\Local\Temp\arduino\sketches\898FEF7A75B3FB13B3B7BCE41BF8A1F5\sketch\appSpecific.cpp.o:(.literal._Z9doAppPingv+0x8): undefined reference to `relayPin'
/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: 
\AppData\Local\Temp\arduino\sketches\898FEF7A75B3FB13B3B7BCE41BF8A1F5\sketch\appSpecific.cpp.o:(.literal._Z9doAppPingv+0xc): undefined reference to `relayMode'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

s60sc, 2024-09-26:
download latest and clean instal

deanfourie1, 2024-09-27:
Thanks that worked also using esp v 2.xx.xx

closed 2024-10-05 by s60sc