s60sc/ESP32-CAM_MJPEG2SD #489: Startup Failure: Check SD card inserted

deanfourie1, 2024-09-27:
So, i've looked though all issues etc, I still cannot find a solution to this.

- I've selected the right board in the appglobals, as far as I can see,
- I've attempted formatting the SD in different formats etc.

Still, I cannot get past this error, "Startup Failure: Check SD card inserted".

Any ideas? My appGlobals.h is as follows.

```
// User's ESP32 cam board
#if defined(CONFIG_IDF_TARGET_ESP32)
//#define CAMERA_MODEL_AI_THINKER 
//#define CAMERA_MODEL_WROVER_KIT 
//#define CAMERA_MODEL_ESP_EYE 
//#define CAMERA_MODEL_M5STACK_PSRAM 
//#define CAMERA_MODEL_M5STACK_V2_PSRAM 
//#define CAMERA_MODEL_M5STACK_WIDE 
//#define CAMERA_MODEL_M5STACK_ESP32CAM
//#define CAMERA_MODEL_M5STACK_UNITCAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL 
#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_TTGO_T_CAMERA_PLUS
//#define AUXILIARY

// User's ESP32S3 cam board
//#elif defined(CONFIG_IDF_TARGET_ESP32S3)
//#define CAMERA_MODEL_FREENOVE_ESP32S3_CAM
//#define CAMERA_MODEL_XIAO_ESP32S3 
//#define CAMERA_MODEL_NEW_ESPS3_RE1_0
//#define CAMERA_MODEL_M5STACK_CAMS3_UNIT
//#define CAMERA_MODEL_ESP32S3_EYE 
//#define CAMERA_MODEL_ESP32S3_CAM_LCD
//#define CAMERA_MODEL_DFRobot_FireBeetle2_ESP32S3
//#define CAMERA_MODEL_DFRobot_Romeo_ESP32S32
//#define AUXILIARY
#endif
```

Thanks

rjsachse, 2024-09-28:
do you have a picture of the board?

deanfourie1, 2024-09-28:
> do you have a picture of the board?

Sure, is this one exactly.

[image]

rjsachse, 2024-09-28:
you have the wrong board defined. try  `CAMERA_MODEL_AI_THINKER`

deanfourie1, 2024-09-28:
Sorry I think I actually copied the wrong config, I initially tested with AI Thinker board selected and I get the same result.

I tried with esp32cam board type as test and this must have been the config I pasted above.

I will test and confirm but I am 100% sure I am getting the same result with AI Thinker board selected.

deanfourie1, 2024-09-28:
so I just tried with the following and I still get the same error.

```
// User's ESP32 cam board
#if defined(CONFIG_IDF_TARGET_ESP32)
#define CAMERA_MODEL_AI_THINKER 
//#define CAMERA_MODEL_WROVER_KIT 
//#define CAMERA_MODEL_ESP_EYE 
//#define CAMERA_MODEL_M5STACK_PSRAM 
//#define CAMERA_MODEL_M5STACK_V2_PSRAM 
//#define CAMERA_MODEL_M5STACK_WIDE 
//#define CAMERA_MODEL_M5STACK_ESP32CAM
//#define CAMERA_MODEL_M5STACK_UNITCAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL 
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_TTGO_T_CAMERA_PLUS
//#define AUXILIARY

// User's ESP32S3 cam board
//#elif defined(CONFIG_IDF_TARGET_ESP32S3)
//
//#define CAMERA_MODEL_FREENOVE_ESP32S3_CAM
//#define CAMERA_MODEL_XIAO_ESP32S3 
//#define CAMERA_MODEL_NEW_ESPS3_RE1_0
//#define CAMERA_MODEL_M5STACK_CAMS3_UNIT
//#define CAMERA_MODEL_ESP32S3_EYE 
//#define CAMERA_MODEL_ESP32S3_CAM_LCD
//#define CAMERA_MODEL_DFRobot_FireBeetle2_ESP32S3
//#define CAMERA_MODEL_DFRobot_Romeo_ESP32S3
//#define AUXILIARY
#endif
```

deanfourie1, 2024-09-28:
Also have cloned latest version and compiled using 3.0.5 esp32

deanfourie1, 2024-09-28:
This is the SD I'm using.

[image]

s60sc, 2024-09-28:
if that is a 256gb its not supported by esp. Max is 32gb

deanfourie1, 2024-09-28:
Ok now im getting this 

I guess these are crappy fake boards.

```
[00:00:04.096 startWebServer] Starting web server on port: 80
[00:00:04.096 startWebServer] Remote server certificates not checked
[00:00:04.107 WARN setup] Startup Failure: Insufficient PSRAM for app 
[00:00:04.107 loop] =============== Total tasks: 12 ===============

[00:00:09.094 resetWatchDog] WatchDog started using task: ping
```

deanfourie1, 2024-09-28:
It does state where I bought my board that it has a 4MB PSRAM

s60sc, 2024-09-28:
the sd is checked first and passes with 8gb then checks psram and fails due to scam board

rjsachse, 2024-10-05:
here is a quick way to check the size of psram

```
#include <esp_heap_caps.h>

void setup() {
  Serial.begin(115200);
  if (psramInit()) {
    Serial.println("PSRAM initialized successfully!");
    size_t psramSize = ESP.getPsramSize();
    Serial.print("PSRAM Size: ");
    Serial.print(psramSize / 1024);
    Serial.println(" KB");
  } else {
    Serial.println("PSRAM initialization failed!");
  }
}

void loop() {
  // Nothing to do here
}

```

what are the results?

closed 2024-10-05 by s60sc