s60sc/ESP32-CAM_MJPEG2SD #492: Remote mic is always Off

madiglechat, 2024-09-28:
Hello,

Very nice job, thanks for this program.

Unfortunatelly, I have a trouble with the microphone, I can't use it, it's always on OFF, even if I configure it on ON on the browser.
[image]
I use a Xaio ESP32S3 card and I want to change the microphone pins number but it's the same problem, the pin WS stay always with the value 41 even if I change it to 42 like in the picture, same trouble with the SD pin.

Thanks for your help.

s60sc, 2024-09-28:
You cant change the esp mic pins they are fixed in the xaio hardware.
The esp mic is local, the remote mic is the browser which will only work if you have an amp/speaker on the esp

madiglechat, 2024-09-29:
In the file camera_pins.h section xiao ESP32S3 the hardware is well defined:
[image]
But in the browser the I2S WS is 41 and the I2S SD is 42, if I change it, save the config, it always come back to the WS=41 and SD=42, and my records have no sound...

closed 2024-10-05 by s60sc