s60sc/ESP32-CAM_MJPEG2SD #496: LED can't be turned on?

vsolenoid, 2024-09-30:
Sorry for the lame question, but it seems I can't turn the stupid LED on. I went to Edit Config and put Pin 4 for used for lamp. Lamp is activated manually. Then saved. But even if I move the slider under Camera Control nothing happens. I uploaded the Blink example just to check if the LED work and it does. Am I missing something? 
Thanks

s60sc, 2024-09-30:
after entering pin number, click focus away from pin field so that browser registers the input

vsolenoid, 2024-09-30:
Hey mate. Thanks for the fast reply. Unfortunately this didn't helped. I saw it in some other solved issues and tried it, but so far I'm unable to save any setting, even if I get a message [23:33:38.160 savePrefs] Saved preferences
[23:33:38.308 saveConfigVect] Config file saved
I just saw my chatbot info was also not saved.
Let me check if I get you right. I should chose the right setting, then click somewhere on the side, then click "save", right?
If it's that... somehow it's all lost by the time it starts again. And the LED never comes on.
Any other suggestions? Should I check for another SD card?

s60sc, 2024-10-01:
SD card wont be a problem, but I dont have an answer.

vsolenoid, 2024-10-01:
Pity. Just to let you know that I've also tried a different browser (just in case) and the results are the same. Unfortunately I have only one esp32 cam, so I can't test it on another. 
So can I somehow set in the sketch itself the LED to be always on? This would be also super helpful 
Thanks

rjsachse, 2024-10-03:
What board are you using?

s60sc, 2024-10-03:
You need to read the latest readme installation instructions and set
`#define INCLUDE_PERIPH true  // peripherals.cpp (servos, PIR, led etc)`
I'll make it clearer in the next release

vsolenoid, 2024-10-04:
> What board are you using?

ESP32 Dev Module

vsolenoid, 2024-10-04:
> You need to read the latest readme installation instructions and set `#define INCLUDE_PERIPH true // peripherals.cpp (servos, PIR, led etc)` I'll make it clearer in the next release

I've tried, but unfortunately I can't compile the sketch, as it give me Error compiling for board ESP32 Dev Module.

I also ordered a new board. We'll see how it behaves once it's here. But it would be super strange if the same error occurs, as I see the LED slider definitely working in the Max Imagination's video

rjsachse, 2024-10-04:
> I've tried, but unfortunately I can't compile the sketch, as it give me Error compiling for board ESP32 Dev Module.
>
> I also ordered a new board. We'll see how it behaves once it's here. But it would [...]

It looks like you might have the wrong board selected. As far as I know, the ESP32 Dev board does not support a camera. Please ensure you are using the correct hardware, have the correct board selected in the Arduino IDE, and have the correct board specified in your code. Need more information to help out further, please double-check everything as this does not seem like an issue with the code itself.

closed 2024-10-20 by s60sc