s60sc/ESP32-CAM_MJPEG2SD #497: in V10.2.2 can't get stored some params entered in WEB front end (e.g. smtp access setting)

vermoi63, 2024-10-03:
pls help, worked great in last V9...but no longer in V10

-formated SD new
-put new data folder on it
-have smtp feature included
-cleared Browser cache, different browser also chrome, they all worked before
-get positive message...config filse saved - but after reboot it' gone
-also tried the web control delete=1

examples that I do net get stored over reboot: time zone setting and smtp settings
seems params in "access setting" are affected - not all
others like host name also in access setting I get stored
others in other TAB's I also get stored like cam and motion setting

also I do NOT get stored e.g. the "enable email sending" in other Tab

Ideas? 
Kind regards!

s60sc, 2024-10-03:
You need to read the latest readme installation instructions and set 
`#define INCLUDE_SMTP true    // smtp.cpp (email)`
Probably the reason for your other issue.
I'll make it clearer in the next release

vermoi63, 2024-10-03:
Great Tks, shame on me - I replaced alsmost everything with "false" but for "true" I left "...included...".

Now it works fine again
Kind regards!
Michael

vermoi63, 2024-10-03:
fine now

closed 2024-10-03 by vermoi63