s60sc/ESP32-CAM_MJPEG2SD #498: (not really code related): Imrove StatUp of esp when powered by UPS - to often "crash loop detected..."

vermoi63, 2024-10-03:
Some Ideas on this - perhaps solution by software possible?

I'm powering the AI Thinker board with a battery extension pack as a UPS...UPS powering by PV...

Config:
PVmodule with USBout --> Bat. extensionPack with Lion --> USBout --> esp32CAM dev board (including USB part of board)

Behavior:
works fine in general..esp32 running on bat-power and bat can be charged at any time without esp down

BUT:
when the bat goes emty it cuts the power to the esp - also fine
Once I recharge the bat with PV or Powerbank the esp is immediately powered.....and shows "startup failure...loop crash..."
And from this state it never comes up again unattended!!!

Assumption:
in the very eary stage of recharging the bat unfortunately the extension pack immediately provides an USBout  but my not yet be able (due to SoC) to drive the esp fully....this my lead to the coded startup failure:

Solutions?
(sure Issue is that UPS goes up on USB out to eary)

but as the code seems to run already (as crash loop can be reported via WLAN to browser.....the esp is not fully dead - so there may be a software solution for this?'

Ideas:
- putting a delay(...) after the coded print of the "crash loop" and doing an ESP.restart() after 10 mins?
(not sure this code really runs properly until the restart on the esp)...need to try..first test show the restart never happens...

- using a Waveshare "solarmanager" instaed of cheap extension pack as a UPS?...that my behave different on releasing the output voltage after being empty?...will try as I have one..open

- putting the USB part off from the esp32 devboard and hoping that powering the reduced devboard behaves better?.
nee to try....open'
'
- sure making sure the bat nevers runs empty would also be a workaround, but also not really a good solution and perhaps a overkill

- hardware solution that does not release the USB out directly after start of charging?

- or is it simply a too sensitivy code for crash loop?
(though crashed are more stable in V10 but seems also to happen in this particular case)'
so looping the crash loop exit with NO exit and loop until eternaty or until OK a solution?

- others
THIS IS WHY I POST it here!!
Any idea on possible coding after the detection of a "crash loop" the restarts the esp or let wait or or or....after a while....as at least the code still seems to have control on the esp...so there may be something that can let wait and more or less simulate plug off/on or restart from power--supply? (or do something with a gpio in this state to restart in loop until ok....or can sleep be used (but even if a sleep code after print WARN would work I would have nothing to hardware signal the wake up...(as bat can run emty at any time when PV is bad).....pls help!

Thanks for any idea on software-solution (or if required) hardware solution

s60sc, 2024-10-03:
Your power supply should not start up the esp until the battery has sufficient charge - that is why it crash loops, so use a more intelligent supply.
You can reboot from a crash loop remotely by entering <your ip>/control?reset=1 in the browser, but if the problem hasnt been fixed it will just crash again
I'll add a reboot button to the warning in the next release.

vermoi63, 2024-10-03:
Tks for fast reply!

Sure you are right about the root cause - I though as long as the Code has control on the CPU there could be a automatic workaround...examples I wrote...lopping until it works, sleep a while, but how to wake up?....or anything like this

yes the esp is powered up to early - BUT code is running already!
so I need something that retries proper restart automatically as the device shall be unattended and should go by itself ON.

It is about 10 mins - 30 mins I need to wait until net try, that it works (which is also show by the fact that in this un-proper state a RST is ok (no need to cut power OFF

Sure I want to solve something at the wrong place, but such a UPS is very cheap...

- perhaps someone knows a simple HW solution? to wait for charge...
- or something with sleep?
- or looping in an intervall

Perhaps I can somehow use your idea that in this state the code is still under control and a external WEB call can force a retry...but this requires something externally....hmmmmm?

Thanks, and I'm still interested in ideas to solve my problem even at the wrong place.

-as mentioned perhaps putting the USB/UART part away from the Dev.board may help as it also sucks current (I did not yet try) OR
- instead the very cheap Bat Extension Pack I will try the waveshare Solarmanager that may have a different output handling and set USB OUT = ON only if a minimum of LION Bat. charging Voltage is guaranteed.

- also I will contact AZdelivery why the extension pack immediately turns the Ouput ON as sonn as something provides a charging voltage......this would be the real root cause, buit I don's assume someone does a re-engineering (perhaps they have an idea for a small HW extension a king of a battery guard.

Thanks again for sharing all you precious work with us!

vermoi63, 2024-10-03:
add: idea could be......adding a retry-count (param1) for a automatical restart after a while/time (param2) that can be configure!

my delay(..) and than ESP.restart after your code did not work - I assume simply becaue it is blocking, so perhaps it work non-blocking...

this could be an idea for UNATTENDED devices somewhere in the field with self-healing :-)

rjsachse, 2024-10-05:
In Arduino ide what core do you have set for Arduino runs on core:? and Events runs on core:?

vermoi63, 2024-10-05:
Final: can't solve this HW issue by SW...
but can use the WEB request for reset control and the fact that in advance to powerdown code get brownout where I can at least react on

Greatings!

closed 2024-10-05 by vermoi63