s60sc/ESP32-CAM_MJPEG2SD #499: "ESP32-CAM Issue: Reappearing '19700101' GHOST Folder and Auto SD Delete Failure During Long PIR-Triggered Recordings, Causing Corrupt Files and SD Card Overflow"

ChandrachurMaitra, 2024-10-03:
Hi sir, I'm having a new problem, after accessing the folder containing the files (through 192.168.4.1) I can see there is a folder named 19700101, in which there are sometimes files and sometimes it's empty BUT STILL EATS SIGNIFICANT STORAGE due to CORRUPT FILES (I guess). When using the "Auto sd delete" feature then the 17700101  folder gets deleted with the other oldest folders (except the current folder) but strangely the 19700101 folder reappears again for some reason, I did not even restart the esp.

**The second problem I noticed is that let's say I have set the auto SD to delete as--> **"when 1Gb will be left on the SD, then the Auto SD delete"** feature will trigger. As you might know, I'm using a PIR, and based on the PIR set delay time, it will remain HIGH, and this time, the camera will keep recording.  

**And here the real problem starts (Overview):** 
Let's say I have 1.2 GB left (using 4GB SanDisk) on your SD card and at this time the PIR starts (set to Repeatable Trigger Mode ) so that the camera starts recording, and let's say the PIR triggered again before the time delay finished, so the pir time delay gets RESET and counts from the beginning and thus the recording stops after (let's say) 20-25 minutes later (during the hot day time) and at that time the size of the recording become huge (800 MB almost), but this is not the real problem, the real problem explained below, it's just the overview.

**_Real Problem:_**
**So when the recording starts, the SD card has only 500 MB left and the "AUTO SD DELETE" set to let's say at "300MB", so as you can see when the PIR GETS TRIGGERED it does almost 700mb continuous recording and I HAVE SEEN FROM THE LOGS (& also from the serial monitor), **THE "AUTO SD DELETE" FEATURE DOSN'T DELETE ANY FILES UNLESS THE CURRENT RECORDING STOPS AND SAVED**, SO BASED ON THIS SCENARIO YOU CAN SEE AND UNDERSTAND THE MEMORY CARD GETS COMPLETELY FULL AND thus THE WEB PAGE (192.168.4.1) GETS COMPLETELY STUCK, NO MATTER WHAT I DO NEXT, THE ESP NEVER ABLE TO DELETE THOSE FILES UNTIL I DO MANUAL FORMAT THE SD CARD & **needless to say in this scenario the http://192.168.4.1/control?deldata=1 method also doesn't work**.

**If the ESP AI THINKER even starts by luck then there already creates some CORRUPT INVISIBLE files because of this and those files are wired because you can't see them even if you plug the SD card into your computer and CHECK THE "SHOW HIDDEN FILES" box of your PC's FILE EXPLORER** but it consumes significant amount of SD storage based on corrupted file size, and the only way to receive back the full storage you have to format the sd card manually using a PC and a card reader. (Please see the first photo if needed)

**LESS IMPORTANT BUT STILL TELLING:** 
**When this incident happens I reset the board by the button & when the ESP CAM AI THINKER starts and gives Access Point, then the name strangely changes from THE DEFAULT name (ESP32_MPEG2SD) to "ESP_F13A15" (please see the 2nd photo if needed)**

 And some days ago, I faced the same problem so when I took the SD card and inserted it into my PC I saw a 799 MB THUMBNAIL-LESS video and when I tried to start the video in VLC then the VLC gave an error saying **"Because this file index is broken or missing**, seeking will not work correctly. VLC won't repair your file but can temporarily fix this problem by building an index in memory." (Please see 3rd photo if needed)

**Here is the photo of the problems: **
[image]

[image]

[image]

s60sc, 2024-10-03:
19700101 is beacuse the esp has not received a time signal.
The app is not designed for continuous recording on SD you should stream to an NVR

ChandrachurMaitra, 2024-10-04:
Ohh IC, now I understand the 179700101 is Unix Epoch, and I think NVR is the best way to go.

closed 2024-10-05 by s60sc