s60sc/ESP32-CAM_MJPEG2SD #501: v10.2.2 PIR usage unclear but wire-interrupt works pin4 AI thinker

vermoi63, 2024-10-04:
I'm confused,
I use PIN4 for PIR....the PIR is one of those (AM...) that go HIGH for 2 sec if motion is detected.

If I connect it to the AI Thinker and PIR goes up I see the LED also PIN going on...so the PIR works.
But no detection in the program!!!
But if I interrupt the signal wire briefly the motion and smtp works.

Could it be that this sensor is not supported in logic?

vermoi63, 2024-10-04:
:24:44.317 setupLamp] Setup PWM Lamp Led on pin 33
[16:24:44.322 resetWatchDog] WatchDog started using task: ping
[16:24:44.328 getLocalNTP] Using NTP server: pool.ntp.org
[18:24:44.339 showLocalTime] Got current time from NTP: 04/10/2024 18:24:44 with tz: CET-1CEST,M3.5.0,M10.5.0/3
[18:24:44.366 updateAppStatus] Disabling motion detection
[18:24:44.517 prepRecording] To record new AVI, do one of:
[18:24:44.518 prepRecording] - press Start Recording on web page
[18:24:44.518 prepRecording] - attach PIR to pin 4
[18:24:44.528 prepRecording] - raise pin 4 to 3.3V

rjsachse, 2024-10-05:
The issue with the ESP32-CAM is that many of the pins are already pulled high. The best option would be to use RX2 (GPIO 16) since pins 12, 13, 14, 15, 2, 4, 1, and 3 are not suitable. On the ESP32-CAM, only pin 16 will work with a PIR sensor that pulls high. However, you need to change the code and make a hardware modification to pull the signal low, potentially using pins 12, 14, or 4. Note that pin 16 is not an interrupt pin, so you would need to check if the code is polling the pin or using interrupts.

For a hardware modification, you would need to use a transistor to invert the signal and adjust the code accordingly. The ESP32-CAM has very limited pins available for use.

vermoi63, 2024-10-05:
Many Tks for the precise explanation.

pin16 for PIR  causes code "crash loop,
--> so I tried pin12 in the AI Thinker....using a jumper wire and touching v3.3 works fine in the app,
BUT using the AM312 PIR (that on the esp8266 can be used without any pulldown etc.....does not trigger the code
(AM312 states it can be used w/o resistor and is HIGH active for 2sec or more if motion continues)

Could it be that pin12 again is not "normal" (as directly to v3,3 triggers the code)?
(late I will try something with pulldown and today me freenove will be delivered (seem everything advanced is difficult with AI Thinker...which is also stated by developers for a good reason it seems)

(just though I found comment where people manages PIR on AI Thinker.....but not for me or not easy with AM312?

Kind regards & mny Tks!

s60sc, 2024-10-05:
AI Overview
Learn more
…
The AM312 PIR sensor has two trigger modes: L (non-repeatable) and H (repeatable): 
L: Outputs a high level when it detects a human body, but only once 
H: Outputs a high level after detecting a human body, and continues to output high until the delay period ends and the person leaves the sensing range

use H mode

vermoi63, 2024-10-05:
Gents, I'm sorry: I had a very stupid mistake - powering PIR with 5V was the issue:

Now AI Thinker works great with PIR=AM312 (continuous mode) / esp PIN 12 / Vpir=3,3

(reading that pin12 should be used as input I need to see if there is a downside using it for the pir.)
(will check 3/4/13 now as I need 12 for relay)

Tks to u

vermoi63, 2024-10-05:
Final Solution:
- AI Thinker Board (without USB board) powered with Bat. Extension Pack/Lion)
-PIR PIN 4 (U0R)
-PIR Sensor AM312...supply with 3,3 and a 4.7k pulldown

Works really great incl, sensitivity, smtp, recording etc.

Greatings!

closed 2024-10-05 by vermoi63