s60sc/ESP32-CAM_MJPEG2SD #503: Error: disableRangeSlider is not defined

limengdu, 2024-10-08:
I used the latest version of the code with XIAO ESP32S3 Sense, esp-arduino version 3.0.5. compiling and uploading went very smoothly, no problems. But when I use Chrome, I encounter Error: disableRangeSlider is not defined when I open the web page. then I can't click on anything to get the screen.
[image]

s60sc, 2024-10-08:
you may have an old common.js cached, otherwise ota latest

limengdu, 2024-10-09:
Thanks for your help, I updated the data folder and the programme runs smoothly. But I can never get the sound of the recorded video on XIAO. I don't know where else I'm missing.
[image]
[image]

s60sc, 2024-10-09:
pins numbers transposed, a corrected version of audio.cpp uploaded

closed 2024-10-10 by limengdu