s60sc/ESP32-CAM_MJPEG2SD #505: PIR implemantation HOW best?

vermoi63, 2024-10-09:
Dear developers,
can't get PIR detection running....
thought I have it on AI Thinker/FreeNove, but it was still the CAM detecting, so here are my questions:

1) when having a 
"PIR PIN defined" and
"use PIR for detection" = ON
I was thinking no CAM detection would run,
but the CAM still detects even if "motion detection" = OFF........what is the idea behind?

2) What settings are to be done to get pure PIR detection?
define (PIN for PIR), (use PIR for det=ON).........others eg motion detection ON/OFF, Sensitivity, duration...etc?

3) What is the idea behind your implementation: a) CAM OR PIR detect?, or CAM AND PIR detect or...other idea?

4) is there any OTHER dependency to get PIR sending a smtp (besides pure smtp setting)?'
eg when it is night?, or when image is pretty dark (so do you read the content)?

5) What is a good PIN for the PIR sensor an FreeNove esp32-S3 CAM?
(tried a lot with defining them and putting it on v3,3 but either crash loop after defining or NO effect to trigger something...)

Pls help, I spent nights with PIR and I was all wrong when I though it worked - it still was the CAM detecting
...(I must admit, I did not read the code so far)

Version = 10.2.2 / Board AI Thinker &Freenove / IDE Boardman. V3.0.5

s60sc, 2024-10-10:
pir & cam motion are independent and can each be switched off

vermoi63, 2024-10-10:
in V 10.2.2 definitely still the CAM triggers the smtp sending mail even when
- motion detection=OFF and
- use OIR for detec = ON....

I'm confused as it is always the CAM Motion....even there is nothing happening on the PIR Pin

s60sc, 2024-10-10:
there is an issue with pir and smtp - you may be 1st person to try that combo. will investigate

vermoi63, 2024-10-10:
oh, great thanks to investigate....

s60sc, 2024-10-10:
the pir/smtp issue fixed in v10.4.1

vermoi63, 2024-10-11:
GREAT THANKS TO YOU!

vermoi63, 2024-10-11:
just did test....
PIR detection-->smtp triggers great now !

(test: App v10.4.1 / AI Thinker..32, / pirpin 12 / smtp google / Sensor AM312
*needed to change peripherals.cpp to INPUT only instead of PULL...xy as this sensor has already signal out 3,3 and supply with 3,3)

Idea: could be an idea to support different sensors by config-parameter in WEB if for PIR a PULLDOWN is required or only INPUT
...AM312 = Input (only)
..HC SR 501 = input pulldown
...RCWL0516 = input pulldown

GREAT THANKS again!

closed 2024-10-11 by vermoi63