s60sc/ESP32-CAM_MJPEG2SD #508: I2C library issue

rjsachse, 2024-10-13:
Firstly, excellent work once again. I apologize for only now getting around to testing the I2C components and encountering an issue with the inclusion of SSD1306Wire.h. It appears to be intended for Arduino rather than ESP32. Below is the error message.

```
c:\Users\rohan\OneDrive\Documents\libraries\SSD1306wire\src\SSD1306wire.cpp:1:10: fatal error: avr/pgmspace.h: No such file or directory
    1 | #include <avr/pgmspace.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

s60sc, 2024-10-13:
https://github.com/ThingPulse/esp8266-oled-ssd1306
ill add a comment

rjsachse, 2024-10-13:
Thanks found a few libraries with SSD1306wire.h

closed 2024-10-13 by rjsachse