s60sc/ESP32-CAM_MJPEG2SD #509: Question about CAMERA_MODEL_NEW_ESPS3_RE1_0

shm4r7, 2024-10-15:
Hi @s60sc . As there is no discussions enabled for this repository I am trying to contact via issue. 
This is the only repository that I've seen has integration with that "no-name aliexpress board" :) 
I was first playing with ESP32-CAM with some success then I tried to use the same code implementation for `ESPS3 RE 1.1` board and it have not worked and I can't find the reason :(

`ESP.getPsramSize();` returns value of 4MB.

`esp_camera_init(&config);` works well, it returns `ESP_OK`. 
Then when I call `esp_camera_fb_get();` it does not work, it hits timeout and returns `nullptr`.

My question is if you might know what could be the reason / if you have any recollection of such a result? 
I want to add that installing code from this repository results in correct work, camera captures frames without any problems.

s60sc, 2024-10-16:
dont quite understand - are you writing your own code rather than using the app

shm4r7, 2024-10-16:
yes, I was writing my own code before I found this app.
As I have written this is the only repository i found someone has configuration for that `ESPS3` board.
I hoped you might have encountered similar problem where camera initialization is OK and then when trying to get frame buffer it does not work -  I tried to find differences but could not spot any.

s60sc, 2024-10-16:
I've not had that issue, there could be all kinds of reasons both hardware and software. 
https://github.com/espressif/esp32-camera/issues/155

closed 2024-10-16 by shm4r7