s60sc/ESP32-CAM_MJPEG2SD #510: Enabling cam seems to affect SD card access

paulmaz1955, 2024-10-16:
First - thanks for this project!

Setting INCLUDE_AUDIO true causes read errors - specifically failure to load /data/common.js and /data/MJPEG2SD.htm. Checking the SD card for errors does not help. The cam web page also does not come up (cannot find uri). Setting INCLUDE_AUDIO back to false clears this up.

I'm using the current version 10.4.2. This happens whether or not a mic (INMP441) is attached. Using an ESP32-CAM.

I can get things working with ESP32-CAM_Audio (https://github.com/spawn451/ESP32-CAM_Audio) using the same hardware.

Any advice regarding where to look for conflicts would be appreciated. If I can provide more useful information please let me know.

Thanks -

s60sc, 2024-10-16:
please provide the startup log

s60sc, 2024-10-16:
If you wired up the mic as given in https://github.com/spawn451/ESP32-CAM_Audio you are using the same pins as the sd card

paulmaz1955, 2024-10-17:
So I am. Should have looked at the pinouts and warnings more closely.
Are any pins besides GPIO16 available that will let me use a mic and the SD card at the same time? I read that 2 and 3 can be used after flashing but that hasn't worked for me.

s60sc, 2024-10-17:
Pin 16 is used by psram. Mic is best used with esp32s3 as per readme

closed 2024-10-20 by s60sc