s60sc/ESP32-CAM_MJPEG2SD #513: Lite version for beginners

sakana1110, 2024-10-29:
Sorry to bother you. I’d like to learn how to use the ESP32-S3 to record video and save it to an SD card. The version you wrote has many other features (which I admire), but it’s a bit hard for me to understand. Could I have a simplified version that focuses solely on recording and saving to the SD card for easier learning? Thank you.

s60sc, 2024-10-29:
The simpler version is already there. Delete all the optional files listed in `appGlobals.h` under:
```
/***************************************************************
  Optional features NOT included by default to reduce heap use 
  To include a particular feature, change false to true
***************************************************************/
```

sakana1110, 2024-10-31:
Thank you for your answer, but I want to write a project myself but I don’t know much about recording and saving to SD card (I tried to write it but failed). I would like to ask you to provide a version that simply records and saves to SD card for easier understanding. , or check out my project [https://github.com/sakana1110/School-project](url) to provide some advice, thank you very much.

closed 2024-11-08 by sakana1110