s60sc/ESP32-CAM_MJPEG2SD #514: Odd things happening

Donbruno100, 2024-11-02:
Hi S60
Firstly I love your work, fantastic...just thought of sending you what I am seeing maybe you can figure this out ,
[image]
as I can't

Everytime I use the Pan and Tilt sliders I get a warning as per picture below however the cam and tilt works perfectly no problem at all, however this warning sign appears is it maybe an internal software line that I didnt insert true/false statement?? this happens evertime a slide the Tilt or Pan sliders
thanks
Regards Bruno

s60sc, 2024-11-02:
thanks - spurious error, corrected mjpeg2sd.htm file uploaded

Donbruno100, 2024-11-03:
You're welcome and thanks to you....

Donbruno100, 2024-11-03:
Consider the issue sorted, works perfectly now...appreciated

naradashen, 2024-11-07:
[00:00:47.926 startPing] Started ping monitoring - On
[00:00:47.929 startWebServer] Starting web server on port: 80
[00:00:47.930 startWebServer] Remote server certificates not checked
[00:00:47.940 resetWatchDog] WatchDog started using task: ping
[00:00:47.940 getLocalNTP] Using NTP server: pool.ntp.org
[00:00:47.951 WARN getLocalNTP] Not yet synced with NTP
[00:00:47.951 WARN wgetFile] Open failed: /data/common.js
[00:00:47.951 WARN setup] Startup Failure: Check SD card inserted
[00:00:47.962 loop] =============== Total tasks: 12 ===============

[00:00:48.568 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm
[00:00:50.717 WARN fileHandler] File does not exist or cannot be opened: /data/MJPEG2SD.htm 

Can anyone explain why this happens?.
I uploaded the data folder to the SD card and somehow it cannot read the files in it.
I ran some test codes to see whether the SD is working. It does.
If anyone knows the issue please help me

closed 2024-12-20 by s60sc